RainerKuemmerle/g2o #405: Question about ParameterSE3Offset.

saifullah3396, 2020-03-30:
Hi,

I am trying to use g2o for implementing bundle adjustment on a depth cam. For that purpose, I am trying to see and understand the edge_se3_pointxyz_depth.cpp file which I deem useful for my case.  For the most part, I can understand the definition of jacobians defined in this file, however, I am unable to understand the presence of the se3 transformation defined by ParameterOffset. My jacobian calculations seem to match yours if I consider this ParameterOffset as identity transform. 

From what I have understood, if I am right, the ParameterOffset is some transformation from the camera local coordinates to sensor coordinates? If that is the case, can you please elaborate if its from camera to sensor or vice verse?

Thank you!

closed 2020-03-30 by saifullah3396

whu-lyh, 2022-01-11:
Hi Saifullah， i have the same confusion about this ParameterSE3Offset, i wander whether you figure it out? thanks

saifullah3396, 2022-01-11:
@whu-lyh I probably did but unfortunately I stopped working on this project a long time ago to be able to help atm.