RainerKuemmerle/g2o #409: error: Not able to locate config.h

athultr1997, 2020-04-17:
I downloaded the source from Github. It was successfully built without any errors. The binary files generated in bin were then placed in `/usr/local/bin`, the library files generated in lib were kept in `/usr/local/lib` and the source code was placed in `/usr/local/include`. Then I copied the data_convert example to a new directory and tried to build it there. Unfortunately, it gave the following errors:
```
[ 50%] Building CXX object CMakeFiles/convert_sba_slam3d.dir/convert_sba_slam3d.o
In file included from /usr/local/include/g2o/stuff/command_args.h:34:0,
                 from /home/athultr/data_convert/convert_sba_slam3d.cpp:31:
/usr/local/include/g2o/stuff/g2o_stuff_api.h:39:10: fatal error: g2o/config.h: No such file or directory
 #include "g2o/config.h"
          ^~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/convert_sba_slam3d.dir/build.make:62: recipe for target 'CMakeFiles/convert_sba_slam3d.dir/convert_sba_slam3d.o' failed
make[2]: *** [CMakeFiles/convert_sba_slam3d.dir/convert_sba_slam3d.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/convert_sba_slam3d.dir/all' failed
make[1]: *** [CMakeFiles/convert_sba_slam3d.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
Any idea why? I tried to run other examples too this way. I was still getting the same error.

YYZZAA, 2020-05-03:
I've met the same problem, I didn't even find a file called 'config.h' in the g2o repository, instead I found 'config.h.in', may be this is a bug ?

RainerKuemmerle, 2020-05-03:
config.h is generated upon build using config.h.in as input (see the
respective rules in CMakeLists.txt). It also should be copied when
installing to the include folder.

Please give more details on what fails an how, for example, the include
directories are defined in your build setup.

YYZZAA, 2020-05-03:
> config.h is generated upon build using config.h.in as input (see the respective rules in CMakeLists.txt). It also should be copied when installing to the include folder. Please give more details on [...]

In fact, I'm modifying a project that was written using the older version of g2o, and my task is to replace g2o with the newest version, and I put the g2o repository in the folder 'Thirdparty'. So in the CMakeLists of my project, I use 'add_subdirectory' instead of 'find_package'

The command output looks like this:

In file included from /home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/Thirdparty/g2o/g2o/types/sim3/sim3.h:31:0,
                 from /home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/include/LineTool.h:9,
                 from /home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/include/Flow.h:12,
                 from /home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/include/KeyFrame.h:29,
                 from /home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/include/MapPoint.h:24,
                 from /home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/include/Frame.h:26,
                 from /home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/src/Frame.cc:21:
/home/yangyoung/LineFlowSLAM/src/tracking_wrapper_lineflow/LineSLAM/Thirdparty/g2o/g2o/types/sim3/../../stuff/misc.h:31:24: fatal error: g2o/config.h: 没有那个文件或目录
compilation terminated.
"没有那个文件或目录"　means "can't find that file or directory"

CMakeLists.txt is as follows:

cmake_minimum_required(VERSION 2.8)
project(Line_SLAM)

#IF (NOT CMAKE_BUILD_TYPE)
#SET(CMAKE_BUILD_TYPE Release)
#ENDIF ()
SET(CMAKE_BUILD_TYPE Debug)
message("Build type: " ${CMAKE_BUILD_TYPE})

set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -Wnonnull")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   -O3 -Wnonnull")

#Check C++11 or C++0x support 
include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if (COMPILER_SUPPORTS_CXX11)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
    add_definitions(-DCOMPILEDWITHC11)
    message(STATUS "Using flag -std=c++11.")
elseif (COMPILER_SUPPORTS_CXX0X)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
    add_definitions(-DCOMPILEDWITHC0X)
    message(STATUS "Using flag -std=c++0x.")
else ()
    message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
endif ()

LIST(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake_modules)

find_package(OpenCV 3 REQUIRED)
find_library(BOOST_SERIALIZATION boost_serialization)
find_package(Eigen3 REQUIRED)

add_subdirectory(Thirdparty/DBoW2)
add_subdirectory(Thirdparty/g2o)

find_package(Pangolin REQUIRED)

find_package(GFLAGS REQUIRED)
find_package(GLOG REQUIRED)

include_directories(
        ${GLOG_INCLUDE}
        ${GFLAGS_INCLUDE}
        ${PROJECT_SOURCE_DIR}
        ${PROJECT_SOURCE_DIR}/include
        ${EIGEN3_INCLUDE_DIR}
        ${Pangolin_INCLUDE_DIRS}
)

set(CMAKE_LIBRARY_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/lib)

file(GLOB_RECURSE SOURCE_INCLUDE_FILES include/*.h)
file(GLOB_RECURSE SOURCE_SRC_FILES src/*.cc src/*.cpp)

#add_subdirectory(Extend) 
#include_directories(${Line3D})

add_library(${PROJECT_NAME} SHARED
        ${SOURCE_INCLUDE_FILES}
        ${SOURCE_SRC_FILES})

target_link_libraries(${PROJECT_NAME}
        ${OpenCV_LIBS}
        ${EIGEN3_LIBS}
        ${Pangolin_LIBRARIES}
        LineSLAMDBoW2
	LineSLAMg2o
        ${GLOG_LIB}
        ${GFLAGS_LIB}
        ${BOOST_SERIALIZATION}
        )
Thanks a lot !

RainerKuemmerle, 2020-05-03:
Try to add
`include_directories(${PROJECT_BINARY_DIR})`
to your CMakeLists
as the g2o CMakeLists should generate config.h in ${PROJECT_BINARY_DIR}

closed 2020-06-28 by RainerKuemmerle

KnineF, 2024-01-12:
I try to replace the Thirdparty version g2o(too old!!!) in ORBSLAM3 with the newest version, and i met the same problem when building ROS examples.

sjulier, 2024-01-12:
I don't know if it's of help but we did a lot of tidying up of ORBSLAM2 for teaching purposes. In particular, I removed the hand hacked versions of the third party software (including g2o) and put in the original source trees and used the cmake external projects infrastructure. This makes it possible to drop in newer versions of code directly, including g2o (although the last time I did this was about en months ago). The code is at:

https://github.com/UCL/COMP0130_22-23_Topic_03