RainerKuemmerle/g2o #410: How to add weights for multi-constraints pose estimation

LounesDev, 2020-04-24:
Hello,

I am dealing with a pose estimation problem solved by Levenberg-Marquardt. I tried to integrate an hybrid residual but I cannot find a proper way to weight my constraints. I have a problem like this :
Residual = alpha * Residual1 + beta * Residual2 (with alpha, beta constraint's weights)
How can I integrate these weights ? Do I just have to multiply each error with the weight in the `computeError()` function ?

Regards,

sjulier, 2020-05-15:
Is residual the error term or the squared error term?

LounesDev, 2020-05-15:
The squared error term.

sjulier, 2020-05-15:
Okay... So as I understand it,
residual1 = e1^2
Residual2 = e2^2

where e1 and e2 are errors.

In terms of just getting the squared error terms out, I think there are two ways you could do it:

1. Redefine your error vector so it has the form [e1;e2]. The information matrix would then be block diagonal with [alpha*sigma1;0;0beta*sigma2] where sigma1 and sigma2 are the information matrices associated with each term.
2. Create two separate edges, one for the Residual1 and the other for Residual2. When the objective function is assembled, it would use the two residuals above

However, are you looking at something more like a mixture model (e.g.,https://april.eecs.umich.edu/pdfs/olson2012rss.pdf)?

songming1995, 2020-11-20:
You can rewrite the residuals to make them be likelihood of the potential matches.  In this way, you can sum them up without considering the weighting issue.

closed 2025-02-02 by github-actions[bot]