RainerKuemmerle/g2o #412: vcpkg version doesn't work with ORB-SLAM2

RollingIsland, 2020-05-13:
I installed g2o with vcpkg successfully but when I tried to build ORB-SLAM2 it went wrong. Has the code been updated in vcpkg? Or what else possible reason for the error shown below?
[image]

mgladkova, 2020-05-15:
You don't need to install g2o separately to run ORB-SLAM2 since it has the source code in the subfolder [Thirdparty](https://github.com/raulmur/ORB_SLAM2/tree/master/Thirdparty/g2o). Just make sure that g2o is compiled (see [build.sh](https://github.com/raulmur/ORB_SLAM2/blob/master/build.sh) script).

RollingIsland, 2020-05-15:
@mgladkova  Thanks for your kind reply!
Well, I wonder whether it's **no need** to install g2o separately, or it's just **wrong** not to use the g2o code in ORB-SLAM2? I'm new to SLAM & g2o library.

mgladkova, 2020-05-15:
In this context I am more inclined to say that it is **highly recommended** to use the g2o version provided by ORB-SLAM2, unless you would like to dig into the code of ORB-SLAM2 and make it compatible with the newest version of g2o.

sjulier, 2020-05-15:
We haven't done it with vcpkg, but under Linux and macOS we regularly change the version shipped with ORBSLAM and CCM-SLAM. There are several reasons, including the fact that the older version of g2o triggers a lot of warnings in Eigen.

Chaning the code just takes a few minutes of work. The main difficulties are:

1. The graph now uses std::unique_ptr for creating the components. If you  look at the examples in g2o you can see how to do it - basically it's the same sets of classes, but they need to be wrappped and moved.
2. ORBSLAM uses a "compressed" version of a subset of the types libraries. If you switch to standard g2o, you just need to change the directories in the include statements.

RollingIsland, 2020-05-15:
@sjulier  Thanks for your reply! I will try as you told.

sjulier, 2020-05-15:
Let me know what happens. I haven't tried vcpkg. It's on my list to try.

closed 2025-02-01 by github-actions[bot]