RainerKuemmerle/g2o #416: Linking error

niebayes, 2020-05-25:
This happens while linking. It seems I missed some shared libraries, but I cannot figure them out.
[100%] Linking CXX executable g2o_curve_fitting
Undefined symbols for architecture x86_64:
  "g2o::HyperGraph::Edge::setId(int)", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::HyperGraph::Edge::resize(unsigned long)", referenced from:
      g2o::BaseUnaryEdge<1, double, CurveFittingVertex>::resize(unsigned long) in g2o_curve_fitting.cc.o
      vtable for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "g2o::SparseOptimizer::setVerbose(bool)", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::SparseOptimizer::setAlgorithm(g2o::OptimizationAlgorithm*)", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::SparseOptimizer::initializeOptimization(int)", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::SparseOptimizer::optimize(int, bool)", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::SparseOptimizer::SparseOptimizer()", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::SparseOptimizer::~SparseOptimizer()", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::resolveCaches()", referenced from:
      vtable for CurveFittingEdge in g2o_curve_fitting.cc.o
      vtable for g2o::BaseUnaryEdge<1, double, CurveFittingVertex> in g2o_curve_fitting.cc.o
      vtable for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::setMeasurementData(double const*)", referenced from:
      vtable for CurveFittingEdge in g2o_curve_fitting.cc.o
      vtable for g2o::BaseUnaryEdge<1, double, CurveFittingVertex> in g2o_curve_fitting.cc.o
      vtable for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::setMeasurementFromState()", referenced from:
      vtable for CurveFittingEdge in g2o_curve_fitting.cc.o
      vtable for g2o::BaseUnaryEdge<1, double, CurveFittingVertex> in g2o_curve_fitting.cc.o
      vtable for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::Edge()", referenced from:
      g2o::BaseEdge<1, double>::BaseEdge() in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::~Edge()", referenced from:
      g2o::BaseEdge<1, double>::BaseEdge() in g2o_curve_fitting.cc.o
      g2o::BaseEdge<1, double>::~BaseEdge() in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::updateCache()", referenced from:
      vtable for CurveFittingVertex in g2o_curve_fitting.cc.o
      vtable for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::Vertex()", referenced from:
      g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> >::BaseVertex() in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::~Vertex()", referenced from:
      g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> >::BaseVertex() in g2o_curve_fitting.cc.o
      g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> >::~BaseVertex() in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)", referenced from:
      g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*) in g2o_curve_fitting.cc.o
  "g2o::G2OBatchStatistics::_globalStats", referenced from:
      g2o::G2OBatchStatistics::globalStats() in g2o_curve_fitting.cc.o
  "g2o::get_monotonic_time()", referenced from:
      g2o::BlockSolver<g2o::BlockSolverTraits<3, 1> >::solve() in g2o_curve_fitting.cc.o
      g2o::BlockSolver<g2o::BlockSolverTraits<3, 1> >::computeMarginals(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::__1::vector<std::__1::pair<int, int>, std::__1::allocator<std::__1::pair<int, int> > > const&) in g2o_curve_fitting.cc.o
  "g2o::OptimizationAlgorithmGaussNewton::OptimizationAlgorithmGaussNewton(std::__1::unique_ptr<g2o::Solver, std::__1::default_delete<g2o::Solver> >)", referenced from:
      _main in g2o_curve_fitting.cc.o
  "g2o::Solver::resizeVector(unsigned long)", referenced from:
      g2o::BlockSolver<g2o::BlockSolverTraits<3, 1> >::updateStructure(std::__1::vector<g2o::HyperGraph::Vertex*, std::__1::allocator<g2o::HyperGraph::Vertex*> > const&, std::__1::set<g2o::HyperGraph::Edge*, std::__1::less<g2o::HyperGraph::Edge*>, std::__1::allocator<g2o::HyperGraph::Edge*> > const&) in g2o_curve_fitting.cc.o
      g2o::BlockSolver<g2o::BlockSolverTraits<3, 1> >::resize(int*, int, int*, int, int) in g2o_curve_fitting.cc.o
  "g2o::Solver::Solver()", referenced from:
      g2o::BlockSolverBase::BlockSolverBase() in g2o_curve_fitting.cc.o
  "g2o::Solver::~Solver()", referenced from:
      g2o::BlockSolverBase::~BlockSolverBase() in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::getMeasurementData(double*) const", referenced from:
      vtable for CurveFittingEdge in g2o_curve_fitting.cc.o
      vtable for g2o::BaseUnaryEdge<1, double, CurveFittingVertex> in g2o_curve_fitting.cc.o
      vtable for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::measurementDimension() const", referenced from:
      vtable for CurveFittingEdge in g2o_curve_fitting.cc.o
      vtable for g2o::BaseUnaryEdge<1, double, CurveFittingVertex> in g2o_curve_fitting.cc.o
      vtable for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Edge::clone() const", referenced from:
      vtable for CurveFittingEdge in g2o_curve_fitting.cc.o
      vtable for g2o::BaseUnaryEdge<1, double, CurveFittingVertex> in g2o_curve_fitting.cc.o
      vtable for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::getEstimateData(double*) const", referenced from:
      vtable for CurveFittingVertex in g2o_curve_fitting.cc.o
      vtable for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::estimateDimension() const", referenced from:
      vtable for CurveFittingVertex in g2o_curve_fitting.cc.o
      vtable for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::getMinimalEstimateData(double*) const", referenced from:
      vtable for CurveFittingVertex in g2o_curve_fitting.cc.o
      vtable for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::minimalEstimateDimension() const", referenced from:
      vtable for CurveFittingVertex in g2o_curve_fitting.cc.o
      vtable for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_curve_fitting.cc.o
  "g2o::OptimizableGraph::Vertex::clone() const", referenced from:
      vtable for CurveFittingVertex in g2o_curve_fitting.cc.o
      vtable for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_curve_fitting.cc.o
  "typeinfo for g2o::OptimizableGraph::Edge", referenced from:
      typeinfo for g2o::BaseEdge<1, double> in g2o_curve_fitting.cc.o
  "typeinfo for g2o::OptimizableGraph::Vertex", referenced from:
      typeinfo for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_curve_fitting.cc.o
  "typeinfo for g2o::Solver", referenced from:
      typeinfo for g2o::BlockSolverBase in g2o_curve_fitting.cc.o
ld: symbol(s) not found for architecture x86_64

niebayes, 2020-05-26:
I resolved this by simply putting the FindG2O.cmake file in my project. Seems the system cannot find the package by default.

closed 2020-05-26 by niebayes

shrinivas96, 2023-07-19:
@niebayes Was G2O in the same directory of the project or in some system folders?