RainerKuemmerle/g2o #424: compilation error in ceres/jet.h

Angit16, 2020-06-23:
Hello, 

I am trying to build g2o. I got the folowing error.

`/home/anindya/g2o/EXTERNAL/ceres/jet.h:980:8: error: ‘ScalarBinaryOpTraits’ is not a class template
 struct ScalarBinaryOpTraits<ceres::Jet<T, N>, T, BinaryOp> {
        ^~~~~~~~~~~~~~~~~~~~
/home/anindya/g2o/EXTERNAL/ceres/jet.h:984:58: error: type/value mismatch at argument 3 in template parameter list for ‘template<class BinaryOp, class T, int N> struct Eigen::ScalarBinaryOpTraits’
 struct ScalarBinaryOpTraits<T, ceres::Jet<T, N>, BinaryOp> {
                                                          ^
/home/anindya/g2o/EXTERNAL/ceres/jet.h:984:58: note:   expected a constant of type ‘int’, got ‘BinaryOp’
`

Please let me know if this is a bug or if I am missing something. I have all the dependencies installed in my PC though.

thank you for your response.

regards,
Anindya

RainerKuemmerle, 2020-06-23:
> I am trying to build g2o. I got the following error.

Hi,

can you please provide which version of eigen you have installed?
Furthermore, which OS and compiler (+ version you are using).

Best
Rainer

2020, 19:03:

Angit16, 2020-06-23:
Hello,

OS: ubuntu 16.04.6,
gcc-6.5
eigen3.3 installed.

regards,
Anindya

RainerKuemmerle, 2020-06-23:
Is the error you posted, the full error message?
Seems like newer compilers (like gcc8 on Ubuntu 18.04) are working - as
used on Travis CI.

2020, 21:29:

Angit16, 2020-06-23:
This is the only error and surprisingly I get a huge number of warnings on eigen related stuff. The OS is the same as I mentioned above.

Angit16, 2020-06-23:
Hi,  

I removed eigen and re-installed. Then, cleaned the build directory of g2o and re-built it. It works fine now.

thanks,
Anindya

yanmengting117, 2020-06-28:
hello , i meet the same error, how to deal with it?
[ 90%] Building CXX object g2o/examples/bal/CMakeFiles/bal_example.dir/bal_example.cpp.o
In file included from /home/love208/g2o-master/EXTERNAL/ceres/array_selector.h:38:0,
                 from /home/love208/g2o-master/EXTERNAL/ceres/autodiff.h:147,
                 from /home/love208/g2o-master/g2o/examples/bal/bal_example.cpp:43:
/home/love208/g2o-master/EXTERNAL/ceres/fixed_array.h: In static member function ‘static constexpr bool ceres::internal::FixedArray<T, N, A>::DefaultConstructorIsNonTrivial()’:
/home/love208/g2o-master/EXTERNAL/ceres/fixed_array.h:102:13: error: ‘is_trivially_default_constructible’ is not a member of ‘std’
     return !std::is_trivially_default_constructible<StorageElement>::value;
             ^
/home/love208/g2o-master/EXTERNAL/ceres/fixed_array.h:102:67: error: expected primary-expression before ‘>’ token
     return !std::is_trivially_default_constructible<StorageElement>::value;
                                                                   ^
/home/love208/g2o-master/EXTERNAL/ceres/fixed_array.h:102:68: error: ‘::value’ has not been declared
     return !std::is_trivially_default_constructible<StorageElement>::value;
                                                                    ^
In file included from /home/love208/g2o-master/EXTERNAL/ceres/autodiff.h:152:0,
                 from /home/love208/g2o-master/g2o/examples/bal/bal_example.cpp:43:
/home/love208/g2o-master/EXTERNAL/ceres/jet.h: At global scope:
/home/love208/g2o-master/EXTERNAL/ceres/jet.h:980:8: error: ‘ScalarBinaryOpTraits’ is not a class template
 struct ScalarBinaryOpTraits<ceres::Jet<T, N>, T, BinaryOp> {
        ^
/home/love208/g2o-master/EXTERNAL/ceres/jet.h:984:58: error: type/value mismatch at argument 3 in template parameter list for ‘template<class BinaryOp, class T, int N> struct Eigen::ScalarBinaryOpTraits’
 struct ScalarBinaryOpTraits<T, ceres::Jet<T, N>, BinaryOp> {
                                                          ^
/home/love208/g2o-master/EXTERNAL/ceres/jet.h:984:58: error:   expected a constant of type ‘int’, got ‘BinaryOp’
make[2]: *** [g2o/examples/bal/CMakeFiles/bal_example.dir/bal_example.cpp.o] Error 1
make[1]: *** [g2o/examples/bal/CMakeFiles/bal_example.dir/all] Error 2
make: *** [all] Error 2

thank you

RainerKuemmerle, 2020-06-28:
The type_traits header is included in fixed_array.h which whould be fine if your compiler supports C++-11:
https://en.cppreference.com/w/cpp/types/is_default_constructible

Please provide information about your compiler and its version. Furthermore, which OS.

If you do not need examples, consider deactivating them in the cmake configuration of your build.

zzhh00, 2020-06-28:
Hello,
I meet the similar error as below,
`In file included from /home/hz/hz_projects/Third_party/g2o/EXTERNAL/ceres/autodiff.h:152:0,
                 from /home/hz/hz_projects/Third_party/g2o/g2o/examples/bal/bal_example.cpp:43:
/home/hz/hz_projects/Third_party/g2o/EXTERNAL/ceres/jet.h:980:8: error: ‘ScalarBinaryOpTraits’ is not a class template
 struct ScalarBinaryOpTraits<ceres::Jet<T, N>, T, BinaryOp> {
        ^
/home/hz/hz_projects/Third_party/g2o/EXTERNAL/ceres/jet.h:984:58: error: type/value mismatch at argument 3 in template parameter list for ‘template<class BinaryOp, class T, int N> struct Eigen::ScalarBinaryOpTraits’
 struct ScalarBinaryOpTraits<T, ceres::Jet<T, N>, BinaryOp> {
`

OS: ubuntu 16.04.6,
gcc-5.4
eigen3.2.92 installed.

How can I deal with it?

Regards.

zzhh00, 2020-06-28:
And I remvoe eigen and re-install as Angit16 did,however it does not work.

RainerKuemmerle, 2020-06-28:
As said, consider deactivating the examples in your build configuration.
I cannot test on such an old compiler and OS - it's not worth the effort.
As an option, you may also consider older releases that may still be compatible with your configuration of OS/compiler.

flyduck, 2020-07-07:
> Hi,
> 
> I removed eigen and re-installed. Then, cleaned the build directory of g2o and re-built it. It works fine now.
> 
> thanks,
> Anindya

@Angit16 Hi, could you please help me , I meet the same problem, and I want to know how to remove the eigen and how to cleaned the build directory of g2o ? very thanks !

hanjiangxiaozhu, 2020-07-08:
> Hello,
> I am trying to build g2o. I got the following error.
> /home/anindya/g2o/EXTERNAL/ceres/jet.h:980:8: error: ‘ScalarBinaryOpTraits’ is not a class template struct [...]

Hi, I have solved this problem by annotate the relative code.

LongruiDong, 2020-07-15:
> @Angit16 Hi, could you please help me , I meet the same problem, and I want to know how to remove the eigen and how to cleaned the build directory of g2o ? very thanks !

Because my Eigen version is 3.2.9, So I annotate these code as following, It works for me

```
// Specifying the return type of binary operations between Jets and scalar types
// allows you to perform matrix/array operations with Eigen matrices and arrays
// such as addition, subtraction, multiplication, and division where one Eigen
// matrix/array is of type Jet and the other is a scalar type. This improves
// performance by using the optimized scalar-to-Jet binary operations but
// is only available on Eigen versions >= 3.3
/* NOTE HERE!
template <typename BinaryOp, typename T, int N>
struct ScalarBinaryOpTraits<ceres::Jet<T, N>, T, BinaryOp> {
  typedef ceres::Jet<T, N> ReturnType;
};
template <typename BinaryOp, typename T, int N>
struct ScalarBinaryOpTraits<T, ceres::Jet<T, N>, BinaryOp> {
  typedef ceres::Jet<T, N> ReturnType;
};
*/
```

RainerKuemmerle, 2020-07-25:
Yes, we now require Eigen 3.3 which was first released in Nov. 2016 (enough time has passed).
Changing CMakeLists to make this explicit.

commit 56960c54 (2020-07-25): Force Eigen to be 3.3 or higher

fix #424
closed 2020-07-25 by RainerKuemmerle