RainerKuemmerle/g2o #425: Nothing displayed in g2o_viewer on Ubuntu 18.04

hustchenli, 2020-06-26:
I tried to visualize a g2o-file using g2o_viewer. But nothing displayed. There is no warning or error.
[image]
[image]
The example that I used is from [here](https://github.com/gaoxiang12/slambook2/blob/master/ch10/sphere.g2o).
Is there anything missing? How can I use g2o_viewer to visualize g2o-files? Thanks for your attention.

RainerKuemmerle, 2020-06-28:
Can you provide your g2o config.h? It's located inside your build directory. I would assume that g2o is compiled without visualization support.

hustchenli, 2020-06-28:
@RainerKuemmerle Thanks for your comment. Here is the config.h inside build directory: 
[image]

RainerKuemmerle, 2020-07-25:
Looking at your screenshot of the viewer, there seems to be something else not working.
Even if the types cannot be displayed by the viewer, we should see a coordinate frame.

The error about the constructor does not appear on my system (Ubuntu 20.04).
Which version of libqglviewer do you have?

hustchenli, 2020-07-31:
> Looking at your screenshot of the viewer, there seems to be something else not working.
> Even if the types cannot be displayed by the viewer, we should see a coordinate frame.
> 
> The error about the constructor does not appear on my system (Ubuntu 20.04).
> Which version of libqglviewer do you have?

libQGLViewer-2.7.2 and my system is ubuntu 18.04

closed 2024-02-18 by RainerKuemmerle