RainerKuemmerle/g2o #426: About the multithreading support

XunshanMan, 2020-06-27:
Hello, 
I'm trying to optimize a large graph and find it slow using only one core of the CPU.

I found that for ceres, there is a parameter 'num_threads' to do this. 
I'm wondering if there is a similar support for multithreading in g2o.

Thanks!

RainerKuemmerle, 2020-06-28:
Activate G2O_USE_OPENMP in your cmake configuration.
Additonally, be sure to have a solver (e.g., super nodal cholmod) selected and a BLAS that supports multi threading.

XunshanMan, 2020-06-29:
Hi, @RainerKuemmerle 
Thanks! So glad to hear there are multithreading supports.
I'm a noob for g2o, is there a demo, or materials to learn about its multithreading further?
Anyway, thank you, I'll try to figure it out by myself.

RainerKuemmerle, 2020-06-29:
It's an option in the cmake configuration of the g2o build.

XunshanMan, 2020-06-30:
@RainerKuemmerle  Thanks!

closed 2020-07-25 by RainerKuemmerle