RainerKuemmerle/g2o #434: Edge cross-covariances

MinaHenein, 2020-07-31:
How to consider edge cross-covariances?
I want to implement a sim(3) optimization between two configurations of points and their covariances (points associations are given), where I consider the full covariance matrix associated with the configuration of points, so cross-covariances are not assumed to be zeros, i.e the covariance matrix is not a diagonal matrix.

The way I thought about it, is to stack all the edges into a vector and treat it as a new edge, and supply the full covariance matrix as this edge covariance, however this means that the edge size is variable of the number of points seen at each time step, any other suggestions and/or examples of similar cases to solve a dense optimisation, instead of sparse diagonal covariance optimisation?

sjulier, 2020-09-20:
Can you call computeMarginals on the set of vertices you want to compute the cross correlations between?

MinaHenein, 2020-09-22:
I don't want to compute the cross correlations, I wanted to implement an optimization that considers the full covariance matrix associated with the vertices (including the cross correlations). I was able to do this implementing a  new edge type that considers a single sim(3) vertices and lots of xyz point vertices similar to edge_se3_lotsofxyz that is already implemented. Thanks a lot anyway.

closed 2020-09-22 by MinaHenein