RainerKuemmerle/g2o #436: In types/slam2d/edge_se2_pointxy.cpp, the formulation for analytical Jacobian w.r.t SE2 pose may be incorrect

yiping, 2020-08-10:
@RainerKuemmerle

It looks like the Jacobian w.r.t to SE2 pose (_jacobianOplusXi) is computed by taking the apparent partial derivative of the error w.r.t. x, y and theta. But since this is the SE2 group, the derivative should really be taken w.r.t. the tangent vector. The Jacobian w.r.t. the xy point looks right.

`
    _jacobianOplusXi( 0 , 0 ) = aux_2 ;
    _jacobianOplusXi( 0 , 1 ) = -aux_3 ;
    _jacobianOplusXi( 0 , 2 ) = aux_1*y2-aux_1*y1-aux_3*x2+aux_3*x1 ;
    _jacobianOplusXi( 1 , 0 ) = aux_3 ;
    _jacobianOplusXi( 1 , 1 ) = aux_2 ;
    _jacobianOplusXi( 1 , 2 ) = -aux_3*y2+aux_3*y1-aux_1*x2+aux_1*x1 ;

    _jacobianOplusXj( 0 , 0 ) = aux_1 ;
    _jacobianOplusXj( 0 , 1 ) = aux_3 ;
    _jacobianOplusXj( 1 , 0 ) = -aux_3 ;
    _jacobianOplusXj( 1 , 1 ) = aux_1 ;`

RainerKuemmerle, 2020-08-28:
We have a unit test in place which runs the analytic Jacobian and compares with a numeric evaluation of the Jacobian:
https://github.com/RainerKuemmerle/g2o/blob/ab40785e44f788131f52ec673a499faa44976459/unit_test/slam2d/jacobians_slam2d.cpp#L123

Seems ok to me but a patch / PR is welcome.

yiping, 2020-09-03:
Hi, @RainerKuemmerle

Thanks for pointing out to the unit test. I took a further look into the code base and I think I know why I had the question. 

Initially I looked at the implementation of error Jacobian `linearizeOplus` in edge_se3_pointxyz,  
https://github.com/RainerKuemmerle/g2o/blob/ab40785e44f788131f52ec673a499faa44976459/g2o/types/slam3d/edge_se3_pointxyz.cpp#L91
and the implementation of the `oplus` in vertex_se3
https://github.com/RainerKuemmerle/g2o/blob/8564e1e365d4e719dceca6269a0ba203f7f43dec/g2o/types/slam3d/vertex_se3.h#L105
The implementations adopt standard matrix exponential/logarithm mapping in SE(3). Nice and clear. So when I read the code for edge_se2_pointxy, I assumed it would be the same case. But actually, the `linearizeOplus` implementation in edge_se2_pointxy chose the more computationally-efficient alternative retraction (pseudo-exponential/logarithm), which I didn't realize until very recently.
https://github.com/RainerKuemmerle/g2o/blob/ab40785e44f788131f52ec673a499faa44976459/g2o/types/slam2d/edge_se2_pointxy.cpp#L66

And this is fine as long as the `oplus` implemenation in vertex_se2 is consistent (assume the same retraction strategy), which is the case there.

https://github.com/RainerKuemmerle/g2o/blob/ab40785e44f788131f52ec673a499faa44976459/g2o/types/slam2d/vertex_se2.h#L51   

In the pseudo exponential/log mapping, the `oplus` simplifies to `+`, so the apparent partial derivative just works out fine - and that's where my original confusion comes from. Your implementation is correct, I was just expecting something different when I didn't see all the pieces.  

Please let me know if the above explanation makes sense, and feel free to close this issue.

closed 2020-09-08 by RainerKuemmerle