RainerKuemmerle/g2o #437: Triangulation : optimizer crashes when I set all pose vertices to fixed : Aborted (core dumped)

salahkhan94, 2020-08-12:
I'm attempting to use multiple monocular camera views (with known absolute scale poses) of an object to estimate it's structure via **multi-view triangulation**. However, when I set all pose vertices to fixed, I get the following error `/usr/local/include/g2o/core/block_solver.hpp:64: void g2o::BlockSolver<Traits>::resize(int*, int, int*, int, int) [with Traits = g2o::BlockSolverTraits<6, 3>]: Assertion `_sizePoses > 0 && "allocating with wrong size"' failed.
Aborted (core dumped)
`
There are 10 poses, it doesn't crash when I set at least one pose vertex as as optimizable. I reckon it's probably because of the way i'm defining the solver :     

`    std::unique_ptr<g2o::BlockSolver_6_3::LinearSolverType> linearSolver;
    if (DENSE) {
        linearSolver = g2o::make_unique<g2o::LinearSolverDense<g2o::BlockSolver_6_3::PoseMatrixType>>();
    } else {
        linearSolver = g2o::make_unique<g2o::LinearSolverCholmod<g2o::BlockSolver_6_3::PoseMatrixType>>();
    }

    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(
        g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver))
    );
    optimizer.setAlgorithm(solver);
`

Appreciate any help to resolve this issue

ns15417, 2020-12-16:
@salahkhan94 Hi, I am running ORBSLAM2 and met the same assert info as yours. The answer @RainerKuemmerle post, i am not sure why it solves your problem, you simply change your file to the one Rainer given here? Cuz i want to know the reason so that i can move to ORBSLAM to solve my problem.
Thanks! :)

salahkhan94, 2020-12-16:
No, I had used that file as a reference for writing my triangulation code.
I did not replace it as it is, and as far as i know, Orb slam doesn't do
triangulation this way.

Regards