RainerKuemmerle/g2o #438: How to set vertices of EdgeSE3LotsOfXYZ?

MinaHenein, 2020-08-19:
I initialise a new SE3LotsOfXYZ edge, and when I try to set the first vertex to the SE3 vertex, 
g2o::EdgeSE3LotsOfXYZ * e = new g2o::EdgeSE3LotsOfXYZ();
e->setVertex(0, optimizer.vertex(0));

I run into Segmentation fault (core dumped) error, any idea how to fix this?

sjulier, 2020-08-19:
Have you tried something like gdb to generate a stack trace to see what's crashing and where? Have you ensured that optimizer.vertex(0) returns a valid vertex?

MinaHenein, 2020-08-20:
@sjulier Thanks for the advice, I just had to setSize of the edge as soon as I initialise it, and it now works fine.

However looking at the jacobian implementation in  EdgeSE3LotsOfXYZ::linearizeOplus(), I think line 102
_jacobianOplus[i] = Jj;
should be changed to
_jacobianOplus[i] = Jj.block<3, 3>(index, 0);

The jacobian w.r.t the ith point should be the 3x3 block of the 3nx3 matrix Jj, don't you think so?

MinaHenein, 2020-08-21:
I think I'm wrong, the jacobian w.r.t the ith point should still be of dimension (edgeDimension x 3), so it's the full Jj in this case

sjulier, 2020-08-21:
I haven't used this edge type, so I haven't checked it to be honest.

MinaHenein, 2020-08-24:
All good, I'm pretty sure I was wrong. Thanks anyway

closed 2020-08-24 by MinaHenein