RainerKuemmerle/g2o #441: Can not build on windows

ArtlyStyles, 2020-09-28:
Hi, 

I was trying to install g2o on windows bu running the commend:

    script\install-deps-windows.bat

but got the following error, how can I fix it?

Detecting compiler hash for triplet x64-windows...
Starting package 5/4: lapack-reference:x64-windows
Building package lapack-reference[blas-select,core,noblas]:x64-windows...
Could not locate cached archive: C:\Users\gcong\AppData\Local\vcpkg\archives\5c\5cb8584a6f6113de146642a40a9910751a307e4e.zip
CMake Error at ports/lapack-reference/portfile.cmake:8 (message):
  Can't build lapack-reference if clapack is installed.  Please remove
  clapack:x64-windows, and try to install lapack-reference:x64-windows again.
Call Stack (most recent call first):
  scripts/ports.cmake:79 (include)

Error: Building package lapack-reference:x64-windows failed with: BUILD_FAILED
Please ensure you're using the latest portfiles with `.\vcpkg update`, then
submit an issue at https://github.com/Microsoft/vcpkg/issues including:
  Package: lapack-reference:x64-windows
  Vcpkg version: 2020.06.15-nohash

Additionally, attach any relevant sections from the log files above.

ArtlyStyles, 2020-10-13:
Any one knows?

sjulier, 2020-10-15:
What happens if you remove clapack from the list of packages to install? I guess it's saying that clapack and a dependence of suitesparse are having a barney.

ArtlyStyles, 2020-10-15:
Thank you very much @sjulier 

Could you please let me know how to remove it? I am on windows, and I am only run the *.bat file from the repo.

ArtlyStyles, 2020-10-15:
After intall the dependency and when I run 

cmake -G "Visual Studio 15 2017 Win64" -DG2O_BUILD_APPS=ON -DG2O_BUILD_EXAMPLES=ON -DVCPKG_TARGET_TRIPLET="%VCPKG_DEFAULT_TRIPLET%" -DCMAKE_TOOLCHAIN_FILE="%VCPKG_ROOT_DIR%\scripts\buildsystems\vcpkg.cmake" ..

I got the following error:

 Could not create named generator Visual Studio 15 2017 Win64

Generators
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.
  KDevelop3                    = Generates KDevelop 3 project files.
  KDevelop3 - Unix Makefiles   = Generates KDevelop 3 project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles

How can I fix it?

sjulier, 2020-10-16:
I assume you fixed the bat file (edit it in something like Notepad).

The problem you report is a cmake one. You'll need to debug your system setup. e.g., if I put the error into Google, the very first hit is:

https://stackoverflow.com/questions/46561518/cmake-error-could-not-create-named-generator-visual-studio-15-2017-win64-t

This mentions the fact that there could be multiple versions of cmake on your system.

There were also a lot of other pages suggesting several different causes of the problem and recommending different fixes. I'd recommend you go through them yourself to identify the issue and a workable solution.

ArtlyStyles, 2020-10-16:
I changed the command to use the correct cmake, but still have error:

"c:\program Files\cmake\bin\cmake.exe" -G "Visual Studio 15 2017 Win64" -DG2O_BUILD_APPS=ON -DG2O_BUILD_EXAMPLES=ON -DVCPKG_TARGET_TRIPLET="%VCPKG_DEFAULT_TRIPLET%" -DCMAKE_TOOLCHAIN_FILE="c:\src\vcpkg\scripts\buildsystems\vcpkg.cmake" ..

But now I am get this error:

-- Could NOT find CHOLMOD (missing: CHOLMOD_INCLUDE_DIR CHOLMOD_LIBRARIES)
-- A library with BLAS API not found. Please specify library location.
-- A library with BLAS API not found. Please specify library location.
-- LAPACK requires BLAS.
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (CSPARSE)
  does not match the name of the calling package (CSparse).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake_modules/FindCSparse.cmake:26 (find_package_handle_standard_args)
  C:/src/vcpkg/scripts/buildsystems/vcpkg.cmake:493 (_find_package)
  CMakeLists.txt:120 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find CSPARSE (missing: CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY)
-- Found OpenGL: opengl32
-- Compiling with OpenGL support
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (QGLVIEWER)
  does not match the name of the calling package (QGLViewer).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake_modules/FindQGLViewer.cmake:49 (find_package_handle_standard_args)
  C:/src/vcpkg/scripts/buildsystems/vcpkg.cmake:493 (_find_package)
  CMakeLists.txt:175 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find QGLVIEWER (missing: QGLVIEWER_INCLUDE_DIR QGLVIEWER_LIBRARY)
-- Compiling g2o apps
-- Compiling g2o examples
-- Compiling with MSVC
-- Performing Test HAVE_UNDERSCORE_PREFIXED_BESSEL_FUNCTIONS
-- Performing Test HAVE_UNDERSCORE_PREFIXED_BESSEL_FUNCTIONS - Success
CMake Error at C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find Eigen3 (missing: EIGEN3_INCLUDE_DIR EIGEN3_VERSION_OK)
  (Required is at least version "3.3")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindEigen3.cmake:82 (find_package_handle_standard_args)
  C:/src/vcpkg/scripts/buildsystems/vcpkg.cmake:493 (_find_package)
  CMakeLists.txt:368 (find_package)

RainerKuemmerle, 2020-10-22:
I currently do not have a native Windows machine for testing. I rely on AppVeyor to run builds on Windows which seems to be able to build https://ci.appveyor.com/project/RainerKuemmerle/g2o/branch/master?fullLog=true

From your output above, it seems that CMake cannot find any version of Eigen. Maybe you need to download Eigen and provice a path when invoking CMake (see above console output on AppVeyor)

sjulier, 2020-11-17:
Is there any update on this?

closed 2021-10-09 by RainerKuemmerle