RainerKuemmerle/g2o #442: edges unchanged

wxt-github, 2020-10-10:
Hello, if there are some vertices connected by edges, but some of the edges I want to remain unchanged before and after optimization, what should I do? @RainerKuemmerle

RainerKuemmerle, 2020-10-10:
I guess you mean vertices should not change. Set them to fixed by calling setFixed.
If edges should not be taken into account have a look at the different methods to initialize the optimization.

Cheers
Rainer

wxt-github, 2020-10-12:
Not exactly the same,setFixed means absolutely unchanged before and after optimization, but what I want is that certain vertices change before and after optimization, but the relative position between vertices and vertices remains unchanged.Is there any way to achieve this?

sjulier, 2020-10-15:
There isn't an edge type which ensures equality constraints. You could create an edge type which specifies the relative translation and says the covariance is small (information matrix is very big). However, this tends to introduce large gradients which I think slows the optimiser down.

What's the feasibility of not having a "real" vertex but create a "virtual" one?

Suppose you had two vertices v0 and v1, where v1=v0+dv, where dv is fixed and known. Can you just keep a single vertex v0 in the graph, and wherever you'd reference v1, replace it by v0+dv where dv is, say, a fixed parameter value?

RainerKuemmerle, 2020-10-22:
I would try to address it by using an offset between the two vertices. There are several examples that provide this concept.
For example: https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam3d/edge_se3_pointxyz.cpp
which implements observations to 3D points from a pose whereas an offset would represent the equality between "two" poses. You will only have the vertex once in the optimization but express equality by the offset.

closed 2025-02-01 by github-actions[bot]