RainerKuemmerle/g2o #448: g2o_viewer not working on Mac

gowrijsuria, 2020-10-17:
I had installed g2o using homebrew, but g2o_viewer is not working on Mac.
**-bash: g2o_viewer: command not found**
Kindly look into the issue.

Thank you 
Gowri

sjulier, 2020-10-17:
libQGLViewer is not currently shipped by homebrew on macOS. Therefore, g2o_viewer cannot be built.

I guess the only option would be to build from source. The libQGLViewer website has instructions. I haven't tried myself.

Could you have a go and let us know what you find?

BrendanGlancy, 2020-10-28:
I had an issue installing on Mac. I got this message.  but I'm sure my CLT is up to date. 
Warning: You are using macOS 11.0.

Error: Your CLT does not support macOS 11.0.
It is either outdated or was modified.
Please update your CLT or delete it if no updates are available.

sjulier, 2020-10-28:
What about trying the beta version of the CLT?

https://apple.stackexchange.com/questions/401899/homebrew-your-clt-does-not-support-macos-11-0

BrendanGlancy, 2020-10-29:
That’s what I have

sjulier, 2020-10-29:
Unfortunately I don't have macOS 11.0 beta installed and so I can't check / suggest much. It sounds like it's a transient caused by the new OS but I can't suggest anything other than the generic monitor Xcode (there's now a beta 3), homebrew porting status and / or have a dig yourself to see what test is failing and why.

BrendanGlancy, 2020-10-30:
Okay thank you

closed 2020-10-31 by gowrijsuria