RainerKuemmerle/g2o #460: Build error on Ubuntu 18

adamjstewart, 2020-11-27:
I've tried compiling both g2o 2020-04-10 and the master branch on Ubuntu 18.04 with GCC 7.5.0, but both fail with an internal compiler error:
```console
$ 'cmake' '-G' 'Unix Makefiles' '-DCMAKE_INSTALL_PREFIX:STRING=/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs' '-DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo' '-DCMAKE_INTERPROCEDURAL_OPTIMIZATION:BOOL=OFF' '-DCMAKE_VERBOSE_MAKEFILE:BOOL=ON' '-DCMAKE_INSTALL_RPATH_USE_LINK_PATH:BOOL=OFF' '-DCMAKE_INSTALL_RPATH:STRING=/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/lib64;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/ceres-solver-1.14.0-ivuikydr4ns7sg5p5c7ukbemf7zdlqvw/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/glog-0.4.0-4xsp3d43ivrwsx5ma7e2c2dnysepg63z/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/ncurses-6.2-ucxzuaucbjp4gdqlggyrew5vawocj7nz/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/openssl-1.1.1h-cunuse5mg74lagb6ywsc2nhhhh4ckrf4/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/zlib-1.2.11-dsnnbcqycwsxdaundf4ldd73zntcdygx/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/gflags-2.2.2-jlmyfb352rawc757232in462gzeo726v/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/openblas-0.3.12-t2ymetzabwriavmrtcwpsc5vd77xe4bg/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/freeglut-3.2.1-w7fklzzhovez6h46t53s3r2k5tswr67p/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/inputproto-2.3.2-qj67v2nbpmze6zwmwdqzqkhe3ukf3pf4/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libx11-1.7.0-ye5osnv6bwlzjbhjyly56fuhwiw2joo7/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/kbproto-1.0.7-vkdsatohibdtcvtqqn54qk27qvgzvshj/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxcb-1.14-idn3bfvvhyvyjzg6lnsjm6dpu4uypw6u/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libpthread-stubs-0.4-fkbv6toa7v4cuvz7e6ha7fjhgykmbl4g/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxau-1.0.8-4u4y6ttpxyag2ge4awe6gv6shpq7cksx/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/xproto-7.0.31-h6ez65vxi6mpvgjcn2b3dlrglxdd5urx/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxdmcp-1.1.2-7uglqkvygdskspj7xu7t5kcsvbj2chjs/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libbsd-0.10.0-fm6ezrmzcl6jbmug7cmqxz47iunpdzlw/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/xextproto-7.3.0-lqpsn2lincurzgvp5z4et2ob5yseyoob/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxi-1.7.6-hxa25ym7nowggnkl53ofczfpjmbgdm5l/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxext-1.3.3-scsihl2wb2ovibht43qi7mscreqomfd4/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxfixes-5.0.2-iru4fo6l3wgaem3ld4jk2maoyrjqoivt/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxrandr-1.5.0-l3euzmj7lhjjlr2zjjzevyscpwijjgxo/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxrender-0.9.10-66swmenfuioa5hmvp53ayb7hkpxxpv22/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/renderproto-0.11.1-ez7uopdzromltj2vsmpojag5tatun44n/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/libxxf86vm-1.1.4-p7fthij3ch3oqdp6g6mwxpoccmdrmn3y/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/randrproto-1.5.0-x7tfwtrjpgckrea7qtfuv44remkjlurf/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/suite-sparse-5.8.1-3ewoixkngranfljwk4o37awqog3ylqsc/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/gmp-6.1.2-mgjars6cwckt7ow4t6hrhqbegyi5pehk/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/metis-5.1.0-spd76rx5azp3he3skrhfb7oowixa3ylu/lib;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/mpfr-4.0.2-xp5e2lomynny4xkglf45m4gq4q6ts5j5/lib' '-DCMAKE_PREFIX_PATH:STRING=/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/cmake-3.19.1-lr623o5jdliuc7oyuxr3lqpau57fnhad;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/eigen-3.3.8-3jubervmokrqm7rbnmkmcxw5rbfzki2r;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/suite-sparse-5.8.1-3ewoixkngranfljwk4o37awqog3ylqsc;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/ceres-solver-1.14.0-ivuikydr4ns7sg5p5c7ukbemf7zdlqvw;/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/freeglut-3.2.1-w7fklzzhovez6h46t53s3r2k5tswr67p' '-DBUILD_CSPARSE=OFF' '-DCSPARSE_INCLUDE_DIR=/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/suite-sparse-5.8.1-3ewoixkngranfljwk4o37awqog3ylqsc/include' '-DCSPARSE_LIBRARY=/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/suite-sparse-5.8.1-3ewoixkngranfljwk4o37awqog3ylqsc/lib/libcxsparse.so' '/tmp/gem/spack-stage/spack-stage-g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/spack-src'
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/gem/spack/lib/spack/env/gcc/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/gem/spack/lib/spack/env/gcc/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling on Unix
-- Found Cholmod: /home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/suite-sparse-5.8.1-3ewoixkngranfljwk4o37awqog3ylqsc/include
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- A library with BLAS API found.
-- Looking for cheev_
-- Looking for cheev_ - found
-- A library with LAPACK API found.
-- Found CHOLMOD and its dependencies
-- Found CSparse: /home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/suite-sparse-5.8.1-3ewoixkngranfljwk4o37awqog3ylqsc/include
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so
-- Compiling with OpenGL support
-- Found QGLViewer: /usr/include/QGLViewer
-- Compiling g2o apps
-- Compiling g2o examples
-- Compiling with GCC
-- Found Eigen3: /home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/eigen-3.3.8-3jubervmokrqm7rbnmkmcxw5rbfzki2r/include/eigen3 (Required is at least version "3.3")
CMake Deprecation Warning at g2o/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Generating position indpendent code for slam2d because Qt5 was built with -reduce-relocations
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/gem/spack-stage/spack-stage-g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/spack-build-7qqtpvq
$ make -j4
...
[ 39%] Building CXX object g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/solver_eigen.cpp.o
cd /tmp/gem/spack-stage/spack-stage-g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/spack-build-7qqtpvq/g2o/solvers/eigen && /home/gem/spack/lib/spack/env/gcc/g++ -DUNIX -Dsolver_eigen_EXPORTS -I/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/eigen-3.3.8-3jubervmokrqm7rbnmkmcxw5rbfzki2r/include/eigen3 -I/tmp/gem/spack-stage/spack-stage-g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/spack-src -I/tmp/gem/spack-stage/spack-stage-g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/spack-build-7qqtpvq -Wall -W -O2 -g -DNDEBUG -fPIC -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -std=gnu++11 -o CMakeFiles/solver_eigen.dir/solver_eigen.cpp.o -c /tmp/gem/spack-stage/spack-stage-g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/spack-src/g2o/solvers/eigen/solver_eigen.cpp
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g2o/solvers/csparse/CMakeFiles/solver_csparse.dir/build.make:84: recipe for target 'g2o/solvers/csparse/CMakeFiles/solver_csparse.dir/solver_csparse.cpp.o' failed
[... cut ...]
make[2]: Leaving directory '/tmp/gem/spack-stage/spack-stage-g2o-master-7qqtpvqb3cj7olfdqvfvvaymcuxrdlbs/spack-build-7qqtpvq'
CMakeFiles/Makefile2:2009: recipe for target 'g2o/solvers/csparse/CMakeFiles/solver_csparse.dir/all' failed
make[1]: *** [g2o/solvers/csparse/CMakeFiles/solver_csparse.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
...
```
The build works fine on macOS 10.15.6 with Apple Clang 12.0.0. Any idea what could cause this? Let me know if you need any more debugging info.

sjulier, 2020-11-27:
Both build for me just fine on Ubuntu 18.04 with gcc version:

gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I don't use spack. What did you install so far, and what commands were used?

adamjstewart, 2020-11-27:
I've installed many packages so far:
```console
$ spack find
==> 56 installed packages
-- linux-ubuntu18.04-ivybridge / gcc@7.5.0 ----------------------
autoconf@2.69                cmake@3.19.1    glog@0.4.0            libsigsegv@2.12  libxfixes@5.0.2    mpfr@4.0.2       perl@5.32.0         xcb-proto@1.14.1
autoconf-archive@2019.01.06  diffutils@3.7   gmp@6.1.2             libtool@2.4.6    libxi@1.7.6        ncurses@6.2      pkgconf@1.7.3       xextproto@7.3.0
automake@1.16.2              eigen@3.3.8     inputproto@2.3.2      libx11@1.7.0     libxrandr@1.5.0    openblas@0.3.12  randrproto@1.5.0    xf86vidmodeproto@2.3.1
berkeley-db@18.1.40          fixesproto@5.0  kbproto@1.0.7         libxau@1.0.8     libxrender@0.9.10  opencv@4.2.0     readline@8.0        xproto@7.0.31
boost@1.74.0                 freeglut@3.2.1  libbsd@0.10.0         libxcb@1.14      libxxf86vm@1.1.4   opengl@3.1       renderproto@0.11.1  xrandr@1.5.0
bzip2@1.0.8                  gdbm@1.18.1     libiconv@1.16         libxdmcp@1.1.2   m4@1.4.18          openglu@1.3      suite-sparse@5.8.1  xtrans@1.3.5
ceres-solver@1.14.0          gflags@2.2.2    libpthread-stubs@0.4  libxext@1.3.3    metis@5.1.0        openssl@1.1.1h   util-macros@1.19.1  zlib@1.2.11
```
I just installed `g2o` fine outside of Spack, so this definitely seems like a Spack bug. I'm a Spack developer, and I'm trying to add a package to build `g2o`, so I'm going to try digging a little deeper.

One thing I did notice is that the `g2o` build adds `-msse2 -msse3 -mssse3 -msse4.1 -msse4.2` to the compile command. Is there a way to disable this? Spack injects its own compiler flags which may be interfering with this.

Also, it seems like `g2o` builds its own copies of `ceres`, `csparse`, and `freeglut`. Is it possible to use externally installed packages for this? Spack has packages for all of these, so I don't want to duplicate the installations.

adamjstewart, 2020-11-27:
@sjulier can you try building with `cmake -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo`? I think `Release` builds fine but `RelWithDebInfo` hits the ICE. Still confirming that this fixes things on my side.

adamjstewart, 2020-11-27:
When I try `Release` the build fails much later:
```
[ 95%] Building CXX object g2o/apps/g2o_simulator/CMakeFiles/g2o_simulator2d_application.dir/test_simulator2d.cpp.o
cd /tmp/gem/spack-stage/spack-stage-g2o-20200410_git-ip6uuk3tgcickiwm4y3hu35e2yyhhir2/spack-build-ip6uuk3/g2o/apps/g2o_simulator && /home/gem/spack/lib/spack/env/gcc/g++ -DUNIX -I/home/gem/spack/opt/spack/linux-ubuntu18.04-ivybridge/gcc-7.5.0/eigen-3.3.8-3jubervmokrqm7rbnmkmcxw5rbfzki2r/include/eigen3 -I/tmp/gem/spack-stage/spack-stage-g2o-20200410_git-ip6uuk3tgcickiwm4y3hu35e2yyhhir2/spack-src -I/tmp/gem/spack-stage/spack-stage-g2o-20200410_git-ip6uuk3tgcickiwm4y3hu35e2yyhhir2/spack-build-ip6uuk3 -Wall -W -O3 -DNDEBUG -O3 -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -std=gnu++11 -o CMakeFiles/g2o_simulator2d_application.dir/test_simulator2d.cpp.o -c /tmp/gem/spack-stage/spack-stage-g2o-20200410_git-ip6uuk3tgcickiwm4y3hu35e2yyhhir2/spack-src/g2o/apps/g2o_simulator/test_simulator2d.cpp
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g2o/examples/sba/CMakeFiles/sba_demo.dir/build.make:84: recipe for target 'g2o/examples/sba/CMakeFiles/sba_demo.dir/sba_demo.cpp.o' failed
make[2]: *** [g2o/examples/sba/CMakeFiles/sba_demo.dir/sba_demo.cpp.o] Error 4
make[2]: Leaving directory '/tmp/gem/spack-stage/spack-stage-g2o-20200410_git-ip6uuk3tgcickiwm4y3hu35e2yyhhir2/spack-build-ip6uuk3'
CMakeFiles/Makefile2:2718: recipe for target 'g2o/examples/sba/CMakeFiles/sba_demo.dir/all' failed
make[1]: *** [g2o/examples/sba/CMakeFiles/sba_demo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

sjulier, 2020-11-27:
I'll try to look later this evening.

As regards various cmake options, I just started to look at tidying up some aspects of the cmake files.

Rainer can probably comment on why stuff freeglut is there - I think it originally came from the fact (9-10 years ago!) that some of the stuff was less widely available and supported. There might also be issues with stuff like building on Android which I'm not aware of. (I seem to recall the sse stuff came from one platform which is being supported, but I don't recall the details.)

sjulier, 2020-11-27:
As regards the crashes, I guess you've checked the obvious things like running out of memory? Some of the template expansions can get rather heavy.

adamjstewart, 2020-11-27:
I'm working in a VM with only 4 GB of memory allocated. Could that cause the problem?

sjulier, 2020-11-27:
Might be. My limited knowledge is from reports such as:

https://github.com/kaldi-asr/kaldi/issues/3781

adamjstewart, 2020-11-27:
Thanks @sjulier, I gave my VM a bit more memory and built in serial and finally managed to get things installed! I think this can be closed now.

P.S. If you or any other `g2o` developers would like to be listed as official "maintainers" of Spack's `g2o` build recipe, let me know. You don't have to be Spack experts, it just gives us someone to ping when users report build issues.

closed 2020-11-27 by adamjstewart