RainerKuemmerle/g2o #462: Specifying g2o library path does,'t work

oym1994, 2020-12-03:
Hi,
There are two versions of g2o in my ubuntu20.04 system. One is installed via ros. I want use another version which in built from source code. I specify the absolute path of this library but the generated excutable file is still linked to the ros version lib**.so。
What should I do next to link my project excutable file to the specific version g2o?
Thanks!

Here is the cmakelist code:
`set(G2O_LIBRARIES
    ${PROJECT_SOURCE_DIR}/3rd/g2o/lib/libg2o_core.so
    ${PROJECT_SOURCE_DIR}/3rd/g2o/lib/libg2o_stuff.so
    ${PROJECT_SOURCE_DIR}/3rd/g2o/lib/libg2o_csparse_extension.so
)
target_link_libraries(${PROJECT_NAME}
                      PUBLIC
                      ${G2O_LIBRARIES}
                      Threads::Threads
                      yaml-cpp
                      ${OpenCV_LIBRARIES}
                      ${CSPARSE_LIBRARY}
                      DBoW2
                      ${catkin_LIBRARIES}
                      PRIVATE
                      ${LAPACK_LIBRARIES}
                      )`

RainerKuemmerle, 2020-12-03:
This is a setting for the linker to generate the binary.
You may have to check your LD_LIBRARY_PATH to allow dynamically loading of the libs from your 3rd party folder at runtime.
Investigate the output of `ldd <name-of-your-binary`

oym1994, 2020-12-04:
Thank you! Now I just removed the ros version g2o and it works normally now. Later I will modify according to ur suggestion.

closed 2020-12-04 by RainerKuemmerle