RainerKuemmerle/g2o #470: New a vertex object but should not delete

oym1994, 2020-12-27:
Hi, thanks for your great contribution!
I new a vertex object in a local function, it's like
`
void fun(){
.
.
.
auto v = new g2o::shot_vertex();
.
.
.
delete v;
}
`
When getting out of the function, I will obtain double free error.  But if new a int, it won't. It seems that g2o can release its memory automatically. Is it right?

Best regards!

RainerKuemmerle, 2020-12-27:
This depends a lot on what you implemented in the `...`-parts. For example, if you add the vertex to an optimizer, the pointer is owned by the optimizer taking care of releasing the memory of the vertices/edges that have been added.

oym1994, 2020-12-27:
Got it, thank you!

sjulier, 2020-12-27:
Please note the memory model used in C / C++ are very different from those used in, say, Python or Java. (Typically Python / Java hide most of the details. C / C++ does not to give more finegrained control of memory which in some cases can lead to much higher performance.) If you haven't done so yet, I very strongly recommend that you should spend an hour or so checking stacks and heaps to make sure the differences are clear. (I haven't gone through it fully, but https://www.youtube.com/watch?v=_8-ht2AKyH4 seems like a pretty good introductoin.) Also, I personally only use auto in very limited circumstances. C++ is a strongly typed language, and if you use the type system it will automatically catch a lot of errors for you at compile time.

closed 2020-12-29 by RainerKuemmerle