RainerKuemmerle/g2o #473: Segmentation fault in release build

amin-abouee, 2020-12-30:
I compiled and installed the library on my MacBook with the following configuration: 

- Os: mac osx
- Compiler: Apple clang version 12.0.0 (clang-1200.0.32.28)
- Build Type: Release
- cmake config parameters: default

When I add the g20 to my visual-odometry library to do a geometric bundle adjustment for just two cameras, in **release** build, I am getting the following problem:
`visual_odometry(96053,0x10248be00) malloc: can't allocate region
:*** mach_vm_map(size=9269431078060974080, flags: 40000100) failed (error code=3)
visual_odometry(96053,0x10248be00) malloc: *** set a breakpoint in malloc_error_break to debug
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
`

lldb call stack:
`* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff20340462 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff2036e610 libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff202c1720 libsystem_c.dylib`abort + 120
    frame #3: 0x00007fff20333418 libc++abi.dylib`abort_message + 231
    frame #4: 0x00007fff20324a65 libc++abi.dylib`demangling_terminate_handler() + 242
    frame #5: 0x00007fff2021a9b9 libobjc.A.dylib`_objc_terminate() + 104
    frame #6: 0x00007fff20332847 libc++abi.dylib`std::__terminate(void (*)()) + 8
    frame #7: 0x00007fff20335167 libc++abi.dylib`__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 27
    frame #8: 0x00007fff2033512e libc++abi.dylib`__cxa_throw + 113
    frame #9: 0x00000001036dd009 libg2o_types_sba.dylib`std::__1::__split_buffer<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat>&>::__split_buffer(unsigned long, unsigned long, Eigen::aligned_allocator<g2o::SE3Quat>&) [inlined] Eigen::aligned_allocator<g2o::SE3Quat>::allocate(num=<unavailable>, (null)=<unavailable>) at Memory.h:0 [opt]
    frame #10: 0x00000001036dcfc4 libg2o_types_sba.dylib`std::__1::__split_buffer<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat>&>::__split_buffer(unsigned long, unsigned long, Eigen::aligned_allocator<g2o::SE3Quat>&) [inlined] std::__1::allocator_traits<Eigen::aligned_allocator<g2o::SE3Quat> >::allocate(__n=<unavailable>) at memory:1586 [opt]
    frame #11: 0x00000001036dcfc4 libg2o_types_sba.dylib`std::__1::__split_buffer<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat>&>::__split_buffer(this=0x00007ffeefbfe968, __cap=5466486976, __start=72417430297351329, __a=<unavailable>) at __split_buffer:318 [opt]
    frame #12: 0x00000001036d1e00 libg2o_types_sba.dylib`g2o::BaseVertex<6, g2o::SE3Quat>::push() [inlined] std::__1::__split_buffer<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat>&>::__split_buffer(this=0x00007ffeefbfe968, __cap=<unavailable>, __start=<unavailable>, __a=<unavailable>) at __split_buffer:317:1 [opt]
    frame #13: 0x00000001036d1dfb libg2o_types_sba.dylib`g2o::BaseVertex<6, g2o::SE3Quat>::push() [inlined] void std::__1::vector<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat> >::__push_back_slow_path<g2o::SE3Quat const&>(this=<unavailable>, __x=0x0000000145d09b50) at vector:1623 [opt]
    frame #14: 0x00000001036d1da5 libg2o_types_sba.dylib`g2o::BaseVertex<6, g2o::SE3Quat>::push() [inlined] std::__1::vector<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat> >::push_back(this=<unavailable>, __x=0x0000000145d09b50) at vector:1640 [opt]
    frame #15: 0x00000001036d1da5 libg2o_types_sba.dylib`g2o::BaseVertex<6, g2o::SE3Quat>::push() [inlined] std::__1::stack<g2o::SE3Quat, std::__1::vector<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat> > >::push(this=<unavailable>, __v=0x0000000145d09b50) at stack:199 [opt]
    frame #16: 0x00000001036d1da5 libg2o_types_sba.dylib`g2o::BaseVertex<6, g2o::SE3Quat>::push(this=0x0000000145d09aa0) at base_vertex.h:94 [opt]
    frame #17: 0x0000000106ed1f3c libg2o_core.dylib`g2o::SparseOptimizer::push() at sparse_optimizer.cpp:530:14 [opt]
    frame #18: 0x0000000106ed1f14 libg2o_core.dylib`g2o::SparseOptimizer::push(this=0x00007ffeefbfeba0) at sparse_optimizer.cpp:613 [opt]
    frame #19: 0x0000000106ed7486 libg2o_core.dylib`g2o::OptimizationAlgorithmLevenberg::solve(this=0x0000000145d0a380, iteration=<unavailable>, online=<unavailable>) at optimization_algorithm_levenberg.cpp:95:19 [opt]
    frame #20: 0x0000000106ed09ff libg2o_core.dylib`g2o::SparseOptimizer::optimize(this=0x00007ffeefbfeba0, iterations=10, online=<unavailable>) at sparse_optimizer.cpp:400:28 [opt]
    frame #21: 0x0000000100060433 visual_odometry`BundleAdjustment::twoViewBA(std::__1::shared_ptr<Frame>&, std::__1::shared_ptr<Frame>&, double, std::__1::shared_ptr<Map>&) + 1331`

The code works in both debug and RelWithDebInfo, but in release mode, I am getting this segmentation fault.
It is noticeable that both master and release ver. 20201223 have tested.

Library compile configuration: 

- Build type: Release
- C++: 17
- SIMD optimization: -march=native
- Linked library components: g2o::core g2o::types_sba g2o::solver_structure_only g2o::solver_csparse g2o::solver_cholmod

RainerKuemmerle, 2021-01-02:
Did you compile both g2o and your code with `-march=native`?

Otherwise it would be good to have a minimal example that causes the problem. Ideally one we can run on Travis. I do not have access to OSX otherwise.

amin-abouee, 2021-01-02:
Ahh, Ok. The problem was there. I didn't compile the g20 with -march=native. Solved. Thank you.

closed 2021-01-02 by RainerKuemmerle