RainerKuemmerle/g2o #477: Usage if

skasperski, 2021-01-04:


RainerKuemmerle, 2021-01-04:
I think I noticed that on the AD branch and added in this commit
https://github.com/RainerKuemmerle/g2o/pull/471/commits/d2272d0e20cac24b96bbb8de74632d8b8d58f99d

I'll cherry pick it into master.

skasperski, 2021-01-04:
Still does not work. Ceres is installed to "g2o/EXTERNAL/ceres/...", but include-line is just "EXTERNAL/ceres/..."

gregsensem, 2021-01-04:
same here! does rewind to a previous version help? thanks!

RainerKuemmerle, 2021-01-04:
https://github.com/RainerKuemmerle/g2o/releases/tag/20201223_git should still be fine.
I now understood the issue. My bad.

RainerKuemmerle, 2021-01-04:
Created a branch to tackle the issue. https://github.com/RainerKuemmerle/g2o/tree/move-external

skasperski, 2021-01-04:
Now it seems to work for me. (with the move-external branch)

RainerKuemmerle, 2021-01-04:
Thanks, I'll merge.

commit 7fc9d1cc (2021-01-04): Move EXTERNAL folder under g2o

* simplify handling of include directories in case g2o is installed
* fix #477
closed 2021-01-04 by RainerKuemmerle