RainerKuemmerle/g2o #480: Runtime path error: libg2o_types_slam3d.so

alessandroTorresani, 2021-01-14:
Hello,

I am having issues to link correctly the library libg2o_types_slam3d.so to a simple example. 

I successfully compiled G2O with the default cmake params
`cmake .. && make && make install`

Then, I linked the g2o libraries in the _CmakeList.txt_ file:
[image]

However, this simple code does not work:
[image]

The compilation terminates, but the executable cannot find the **libg2o_types_slam3d.so** library. 
`./main: error while loading shared libraries: libg2o_types_slam3d.so: cannot open shared object file: No such file or directory`

This is the output of the `ldd main`
[image]

However, the **libg2o_types_slam3d.so** is present in the g2o installation folder
`ls /usr/local/lib/`

[image]

Am I doing something wrong when linking the library in the _CmakeList.txt_?

RainerKuemmerle, 2021-01-16:
Can you read the so?
`readelf -Ws /usr/local/lib/libg2o_types_slam3d.so`

vahagnIV, 2021-01-17:
I had a similar issue, try running
    sudo ldconfig

alessandroTorresani, 2021-01-18:
@RainerKuemmerle Yes I suppose I can read it, this is the output (only the first lines):

```
readelf -Ws /usr/local/lib/libg2o_types_slam3d.so

Symbol table '.dynsym' contains 1115 entries:
   Num:    Value          Size Type    Bind   Vis      Ndx Name
     0: 0000000000000000     0 NOTYPE  LOCAL  DEFAULT  UND 
     1: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND _ZNSo3putEc@GLIBCXX_3.4 (2)
     2: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND _ZN3g2o7Factory8instanceEv
     3: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND atan2@GLIBC_2.2.5 (3)
     4: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND _ZN3g2o23HyperGraphElementActionclEPNS_10HyperGraph17HyperGraphElementEPNS0_10ParametersE
     5: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEPKc@GLIBCXX_3.4.21 (4)
     6: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND _ZN3g2o16OptimizableGraph4Edge18setMeasurementDataEPKd
```

alessandroTorresani, 2021-01-18:
@vahagnIV Thank you, `ldconfig` seems to fix the problem above. 

However, I got a segmentation fault when running this code
[image]

The line `optimizer.addParameter(cam_params);` makes the program to crash. Is this still related to the problem I had?

closed 2025-02-01 by github-actions[bot]