RainerKuemmerle/g2o #484: Cannot understand the template parameters of BlockSolverTraits

vahagnIV, 2021-01-17:
I am trying to implement the camera calibration with a planar chessboard pattern following the paper by Zhang. 

As optimizable vertices, I have chosen the camera intrinsic parameters and the positions of the calibration pattern in the frames.

My code looks like this (I have stripped all unimportant details):

    /*!
    * Represents the intrinsic parameters fx, fy, cx, cy, k1, k2, p1, p2, k3
    */
    class VertexCamera : public g2o::BaseVertex<9, Eigen::VectorXd>{
        /*Implementation here*/
        Eigen::Vector2d map(const Eigen::Vector3d & vector); // Projects the 3D point on the image
    };
    typedef g2o::VertexSE3Expmap VertexPatternPose;

    /*!
    * Represents the reprojection error for a single corner
    */
    class EdgeError : public g2o::BaseBinaryEdge<2, Eigen::Vector2d, VertexCamera, g2o::VertexSE3Expmap> {
        // Implemented computeError, read, write
        void setPatternPoint(const Eigen::Vector2d & pattern_point){
            pattern_point_ = pattern_point;
        }

        void computeError() override {
            const Eigen::Vector3d  transformed = dynamic_cast<g2o::VertexSE3Expmap *>(this->vertex(0))->estimate().map(pattern_point_);
           _error = dynamic_cast<VertexCamera *>(this->vertex(1))->map(transformed) - _measurement;
        }
      protected:
         Eigen::Vector2d pattern_point_;
    };

    void Calbirate(const std::vector<std::vector<Eigen::Vector2d>> & corners,
                                const std::vector<std::vector<Eigen::Vector2d>> & pattern_points){
        assert(corners.size() == pattern_points.size());
        g2o::SparseOptimizaer optimizer;
        VertexCamera * camera = new VertexCamera;
        // set Id and estimates
        optimizer.addVertex(camera);

        for (size_t frame_id = 0; frame_id < corners.size(); ++ frame_id){
             assert(corners[frame_id].size() == pattern_points[frame_id].size());
             VertexPatternPose *pose = new VertexPatternPose;
             // set ids and initial estimates
             optimizer.addVertex(pose);
             for ( size_t corner_id  = 0; corner_id < corners[frame_id].size(); ++corner_id){
                     EdgeError *e = new EdgeError();
                     e->setVertex(0, camera);
                     e->setVertex(1, pose);
                     e->setMeasurement(corners[frame_id][corner_id]);
                     e->setPatternPoint(pattern_points[frame_id][corner_id]);
                     // Set id and other parameters
                     optimizer.addEdge(optmizer);
             }            
        }
     }

Now I am trying to initialize the optimizer but cannot figure out what do the template parameters mean. Could not find anything in the g2o.pdf.

    typedef g2o::BlockSolver< g2o::BlockSolverTraits<9, 6> >  BalBlockSolver;
    typedef g2o::LinearSolverCholmod<BalBlockSolver::PoseMatrixType> BalLinearSolver;

    std::unique_ptr<g2o::LinearSolver<BalBlockSolver::PoseMatrixType>> linearSolver;
    auto cholesky = g2o::make_unique<BalLinearSolver>();
    cholesky->setBlockOrdering(true);
    linearSolver = std::move(cholesky);

    g2o::OptimizationAlgorithmLevenberg* optimizationAlgorithm = new g2o::OptimizationAlgorithmLevenberg(
        g2o::make_unique<BalBlockSolver>(std::move(linearSolver)));
    optimizer.setAlgorithm(optimizationAlgorithm);
    optimizer.initializeOptimization();
    optmizer.optimize();

On the last line I get the following error:
     /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:285: void Eigen::PlainObjectBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, 9, 9, 0, 9, 9>; Eigen::Index = long int]: Assertion `(!(RowsAtCompileTime!=Dynamic) || (rows==RowsAtCompileTime)) && (!(ColsAtCompileTime!=Dynamic) || (cols==ColsAtCompileTime)) && (!(RowsAtCompileTime==Dynamic && MaxRowsAtCompileTime!=Dynamic) || (rows<=MaxRowsAtCompileTime)) && (!(ColsAtCompileTime==Dynamic && MaxColsAtCompileTime!=Dynamic) || (cols<=MaxColsAtCompileTime)) && rows>=0 && cols>=0 && "Invalid sizes when resizing a matrix or array."' failed.

which hints to me that I am missing something in the g2o API. At least in the solver initialization. What am I doing wrong?

RainerKuemmerle, 2021-01-18:
If I see correctly you have vertices of dimension 9 and 6.
Since you gave those as dimensions to the block solver, you would have to set `setMarginalized(true)` on the pose vertices for computing the Schur complement.
This might however not be really beneficial here. Instead I would try
`typedef g2o::BlockSolver< g2o::BlockSolverTraits<Eigen::Dynamic, Eigen::Dynamic> >  BalBlockSolver;`

vahagnIV, 2021-01-19:
Thank you so much! 

Is there a text that I can read and figure this out on my own? I mean, I do not understand what the marginalization does and why should I set it to true for computing the Schur complement.