RainerKuemmerle/g2o #489: Eigen assertion fails due to unaligned data structures

alexdewar, 2021-01-27:
I am get the following run time error, which is triggered during the static object creations in types_slam3d.cpp:
```
run_video_slam: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 16; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
```

The help page for Eigen states that this problems is caused by unaligned data structures, but I couldn't figure out where in g2o the root problem lies, though I'm happy to provide any extra info needed!

I am building on Linux with gcc v10.2.0 and eigen v3.3.9.

Here is the call stack:
```
libc.so.6!raise (Unknown Source:0)
libc.so.6!abort (Unknown Source:0)
libc.so.6!__assert_fail_base.cold (Unknown Source:0)
libc.so.6!__assert_fail (Unknown Source:0)
libg2o_types_sba_d.so!Eigen::internal::plain_array<double, 16, 0, 32>::plain_array(Eigen::internal::plain_array<double, 16, 0, 32> * const this) (/usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128)
libg2o_types_sba_d.so!Eigen::DenseStorage<double, 16, 4, 4, 0>::DenseStorage(Eigen::DenseStorage<double, 16, 4, 4, 0> * const this) (/usr/include/eigen3/Eigen/src/Core/DenseStorage.h:187)
libg2o_types_sba_d.so!Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 4, 0, 4, 4> >::PlainObjectBase(Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 4, 0, 4, 4> > * const this) (/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:484)
libg2o_types_slam3d_d.so!Eigen::Matrix<double, 4, 4, 0, 4, 4>::Matrix(Eigen::Matrix<double, 4, 4, 0, 4, 4> * const this) (/usr/include/eigen3/Eigen/src/Core/Matrix.h:259)
libg2o_types_slam3d_d.so!Eigen::Transform<double, 3, 1, 0>::Transform(Eigen::Transform<double, 3, 1, 0> * const this) (/usr/include/eigen3/Eigen/src/Geometry/Transform.h:257)
libg2o_types_slam3d_d.so!g2o::BaseVertex<6, Eigen::Transform<double, 3, 1, 0> >::BaseVertex(g2o::BaseVertex<6, Eigen::Transform<double, 3, 1, 0> > * const this) (/usr/src/debug/g2o-20201223_git/g2o/core/base_vertex.hpp:30)
libg2o_types_slam3d_d.so!g2o::VertexSE3::VertexSE3(g2o::VertexSE3 * const this) (/usr/src/debug/g2o-20201223_git/g2o/types/slam3d/vertex_se3.cpp:43)
libg2o_types_slam3d_d.so!g2o::HyperGraphElementCreator<g2o::VertexSE3>::construct(g2o::HyperGraphElementCreator<g2o::VertexSE3> * const this) (/usr/src/debug/g2o-20201223_git/g2o/core/creators.h:67)
libg2o_core_d.so!g2o::Factory::registerType(g2o::Factory * const this, const std::string & tag, g2o::AbstractHyperGraphElementCreator * c) (/usr/src/debug/g2o-20201223_git/g2o/core/factory.cpp:93)
libg2o_types_slam3d_d.so!g2o::RegisterTypeProxy<g2o::VertexSE3>::RegisterTypeProxy(g2o::RegisterTypeProxy<g2o::VertexSE3> * const this, const std::string & name) (/usr/src/debug/g2o-20201223_git/g2o/core/factory.h:138)
libg2o_types_slam3d_d.so!__static_initialization_and_destruction_0(int __initialize_p, int __priority) (/usr/src/debug/g2o-20201223_git/g2o/types/slam3d/types_slam3d.cpp:37)
libg2o_types_slam3d_d.so!_GLOBAL__sub_I_types_slam3d.cpp(void)() (/usr/src/debug/g2o-20201223_git/g2o/types/slam3d/types_slam3d.cpp:74)
ld-linux-x86-64.so.2!call_init.part (Unknown Source:0)
ld-linux-x86-64.so.2!_dl_init (Unknown Source:0)
ld-linux-x86-64.so.2!_dl_start_user (Unknown Source:0)
[Unknown/Just-In-Time compiled code] (Unknown Source:0)
```

RainerKuemmerle, 2021-01-27:
This seems to be VertexSE3. But maybe it is just this one because it is the first one to be allocated by the factory upon registration.
Are you on a 32bit system?
VertexSE3 should have Eigen's aligned operator defined - I checked on master, though.
Can you compile the unit tests in g2o and run them? Does this also trigger the assert?

meekersX, 2021-01-27:
Assuming this `run_video_slam` is from openvslam, ensure that both openvslam and g2o are compiled with the same `BUILD_WITH_MARCH_NATIVE` flag.

Their [Dockerfile.desktop](https://github.com/xdspacelab/openvslam/blob/master/Dockerfile.desktop) and [troubleshooting guide](https://openvslam.readthedocs.io/en/master/trouble_shooting.html#for-slam) recommend `-DBUILD_WITH_MARCH_NATIVE=OFF`, but I have found it also works if both are built with it `ON`. Troubles arise when g2o is built with a different setting than executables that depend on it.

alexdewar, 2021-01-28:
@meekersX This totally did the trick. Thank you so much! Mea cupla for not reading the guide properly...

@RainerKuemmerle I guess this closes the issue then...?

closed 2021-01-28 by RainerKuemmerle