RainerKuemmerle/g2o #495: Using Saved Pose Graph of ORB_SLAM2 to Do Further Optimizations

chamikasudusinghe, 2021-03-01:
**Issue**

I have modified ORB_SLAM2 (https://github.com/raulmur/ORB_SLAM) to be compatible with the library implementation of g2o (https://github.com/RainerKuemmerle/g2o/releases/tag/20170730_git). The intention behind this was to save the graph using opimizer.save() in the g2o format to do further optimizations with the intention of optimizing the extrinsic camera parameters of the keyframes.

But the resultant g2o file comprises data formats that cannot be used along with ceres-solver (https://ceres-solver.googlesource.com/ceres-solver/+/master/examples/slam/pose_graph_3d/pose_graph_3d.cc) unlike the datasets provided here (https://lucacarlone.mit.edu/datasets/).

Expected data types: **VERTEX_SE3:QUAT ID x y z q_x q_y q_z q_w, EDGE_SE3:QUAT ID_a ID_b x_ab y_ab z_ab q_x_ab q_y_ab q_z_ab q_w_ab**

Following are the types of lines that were present in the saved g2o file from ORB_SLAM2.

**VERTEX_SE3:EXPMAP 0 -5.28081e-09 -5.77454e-09 -7.92324e-09 -4.52662e-11 -4.55318e-11 1.83541e-11 1 
FIX 0
VERTEX_SE3:EXPMAP 1 -0.00693567 -0.035008 -0.0159884 -0.0607977 -0.00406262 -0.0363495 0.99748 
VERTEX_SE3:EXPMAP 2 -0.00524271 -0.0472359 -0.0101736 -0.077399 -0.00429221 -0.0421459 0.9961**
.
.
.
**VERTEX_XYZ 82 0.124216 0.123057 0.340551 
VERTEX_XYZ 84 0.116446 0.124854 0.354567 
VERTEX_XYZ 90 0.094751 -0.128109 0.398485** 

I tried to visualize the g2o file using g2o_viewer and it provided an empty visualization.

**I would be pleased to know whether it is possible to do further optimizations using the saved g2o and the g2o library, or this is the right approach to finetune the extrinsic camera parameters.**

**To Reproduce Please Find the Modification in Optimizer.cc in ORB_SLAM2**

```

void Optimizer::BundleAdjustment(const vector<KeyFrame *> &vpKFs, const vector<MapPoint *> &vpMP,
                                 int nIterations, bool* pbStopFlag, const unsigned long nLoopKF, const bool bRobust)
{
    vector<bool> vbNotIncludedMP;
    vbNotIncludedMP.resize(vpMP.size());

    g2o::SparseOptimizer optimizer;
    g2o::BlockSolver_6_3::LinearSolverType * linearSolver;

    linearSolver = new g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>();

    g2o::BlockSolver_6_3 * solver_ptr = new g2o::BlockSolver_6_3(linearSolver);

    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(solver_ptr);
    optimizer.setAlgorithm(solver);

.
.
.

    // Before Optimize
    std::string filename;
    filename = "before_opt.g2o";
    optimizer.save(filename.c_str());
    cout << endl << "g2o file saved" << endl;

```

**Expected Behavior**

It was expected to use the saved g2o file to do further optimizations to optimize the extrinsic camera parameters.

**Screenshots**

[image]
[image]

**Environment**

- Operating system: Ubuntu 18.04
- g2o version:20170730_git

RainerKuemmerle, 2021-03-06:
The fie in Ceres does only support 2D/3D pose graphs and no types from Bundle Adjustment problems.

Does you file also contain edge types?
The barebone g2o lib managed to load some vertices, 5008 in total. But VertexSE3Expmap does not provide an implementation for visualization. VERTEX_XYZ is removed in master and might not be loaded from the file. You may replace the tag with VERTEX_TRACKXYZ or the same g2o version ORB_SLAM2 used to generate the file.

I do not know the implementations in ORB_SLAM2. If custom types are used as well, those might not be loaded by barebone g2o. You would have to provide types libraries for this.

zeal-up, 2023-12-13:
@RainerKuemmerle Hi～I know it's a long time ago issue, but I still finding a way to provide custom-types dynamic library for g2o_viewer. Should I register custom type in g2o_viewer source code and recompile it?

closed 2025-01-31 by github-actions[bot]