RainerKuemmerle/g2o #496: I ran the curve_fitting.cpp program in the example, and I got an error as follows

RainFromCN, 2021-03-06:
the CMakeLists.txt file is as follows:
```cmake
# find g2o package
find_package( G2O REQUIRED )
include_directories( ${G2O_INCLUDE_DIRS} )

# compile and link test4
add_executable( ch6_test4 test4.cpp )
target_link_libraries( ch6_test4 ${OpenCV_LIBS} )
target_link_libraries( ch6_test4 ${G2O_CORE_LIBRARY} ${G2O_STUFF_LIBRARY} )
```

My curve fitting program is provided by the source code without any changes. When I run it, I get the following error:
```
CMakeFiles/ch6_test4.dir/test4.cpp.o: In function `_GLOBAL__sub_I_data_x':
test4.cpp:(.text.startup+0x90): undefined reference to `g2o_optimization_library_dense'
collect2: error: ld returned 1 exit status
CMakeFiles/ch6_test4.dir/build.make:111: recipe for target 'ch6_test4' failed
make[2]: *** [ch6_test4] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ch6_test4.dir/all' failed
make[1]: *** [CMakeFiles/ch6_test4.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
```

Note that if I only include the header file, and then write an empty main function, then the program can be compiled successfully.

Thank you very much for paying attention to this issue in your busy schedule~~~

RainFromCN, 2021-03-06:
神奇的是我运行高博得代码，竟然能正常工作。。。why？？

RainFromCN, 2021-03-06:
When I remove this line, the program works normally:
```
G2O_USE_OPTIMIZATION_LIBRARY(dense);
```
If I don't use it, is it the default to use the optimization method for sparse matrix to solve the linear equation?

RainerKuemmerle, 2021-03-06:
Consider linking G2O_SOLVER_DENSE in addition to core and stuff.

The macro above ensures that the solver is registered in the factory. But if you do not rely on the factory you may actually skip it.

closed 2021-03-06 by RainerKuemmerle