RainerKuemmerle/g2o #498: base_fixed_sized_edge problem

Airplane5, 2021-03-07:
I am compiling a SLAM framework whose dependecies include g2o.
After installing g2o, there is a problem when compiling SLAM.

`
[ 25%] Built target realm_core
[ 46%] Built target realm_io
[ 55%] Built target realm_densifier_base
[ 70%] Built target realm_ortho
[ 72%] Building CXX object modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/src/visual_slam_factory.cpp.o
In file included from /usr/local/include/g2o/types/sba/edge_project_psi2uv.h:30:0,
                 from /usr/local/include/g2o/types/sba/types_six_dof_expmap.h:38,
                 from /usr/local/include/openvslam/data/keyframe.h:13,
                 from /home/wlh/OpenREALM/modules/realm_vslam/realm_vslam_base/include/realm_vslam_base/open_vslam.h:31,
                 from /home/wlh/OpenREALM/modules/realm_vslam/realm_vslam_base/include/realm_vslam_base/visual_slam_factory.h:31,
                 from /home/wlh/OpenREALM/modules/realm_vslam/realm_vslam_base/src/visual_slam_factory.cpp:21:
/usr/local/include/g2o/core/base_fixed_sized_edge.h:174:32: error: ‘index_sequence’ is not a member of ‘std’
   struct HessianTupleType<std::index_sequence<Ints...>> {
                                ^~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:174:32: error: ‘index_sequence’ is not a member of ‘std’
/usr/local/include/g2o/core/base_fixed_sized_edge.h:174:51: error: expected parameter pack before ‘...’
   struct HessianTupleType<std::index_sequence<Ints...>> {
                                                   ^~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:174:51: error: template argument 1 is invalid
/usr/local/include/g2o/core/base_fixed_sized_edge.h:174:54: error: expected unqualified-id before ‘>’ token
   struct HessianTupleType<std::index_sequence<Ints...>> {
                                                      ^~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:181:38: error: ‘make_index_sequence’ is not a member of ‘std’
       typename HessianTupleType<std::make_index_sequence<_nr_of_vertex_pairs>>::type;
                                      ^~~~~~~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:181:38: error: ‘make_index_sequence’ is not a member of ‘std’
/usr/local/include/g2o/core/base_fixed_sized_edge.h:181:58: error: template argument 1 is invalid
       typename HessianTupleType<std::make_index_sequence<_nr_of_vertex_pairs>>::type;
                                                          ^~~~~~~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:181:16: error: expected nested-name-specifier
       typename HessianTupleType<std::make_index_sequence<_nr_of_vertex_pairs>>::type;
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:183:38: error: ‘make_index_sequence’ is not a member of ‘std’
       typename HessianTupleType<std::make_index_sequence<_nr_of_vertex_pairs>>::typeTransposed;
                                      ^~~~~~~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:183:38: error: ‘make_index_sequence’ is not a member of ‘std’
/usr/local/include/g2o/core/base_fixed_sized_edge.h:183:58: error: template argument 1 is invalid
       typename HessianTupleType<std::make_index_sequence<_nr_of_vertex_pairs>>::typeTransposed;
                                                          ^~~~~~~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:183:16: error: expected nested-name-specifier
       typename HessianTupleType<std::make_index_sequence<_nr_of_vertex_pairs>>::typeTransposed;
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:203:32: error: ‘std::index_sequence’ has not been declared
   bool allVerticesFixedNs(std::index_sequence<Ints...>) const;
                                ^~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:203:46: error: expected ‘,’ or ‘...’ before ‘<’ token
   bool allVerticesFixedNs(std::index_sequence<Ints...>) const;
                                              ^
/usr/local/include/g2o/core/base_fixed_sized_edge.h:208:75: error: ‘std::index_sequence’ has not been declared
  linearizeOplus_allocate(JacobianWorkspace& jacobianWorkspace, std::index_sequence<Ints...>);
                                                                     ^~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:208:89: error: expected ‘,’ or ‘...’ before ‘<’ token
 s_allocate(JacobianWorkspace& jacobianWorkspace, std::index_sequence<Ints...>);
                                                                     ^
/usr/local/include/g2o/core/base_fixed_sized_edge.h:216:30: error: ‘std::index_sequence’ has not been declared
   void linearizeOplusNs(std::index_sequence<Ints...>);
                              ^~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:216:44: error: expected ‘,’ or ‘...’ before ‘<’ token
   void linearizeOplusNs(std::index_sequence<Ints...>);
                                            ^
/usr/local/include/g2o/core/base_fixed_sized_edge.h:240:38: error: ‘std::index_sequence’ has not been declared
                                 std::index_sequence<Ints...>);
                                      ^~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.h:240:52: error: expected ‘,’ or ‘...’ before ‘<’ token
                                 std::index_sequence<Ints...>);
                                                    ^
[... 138 lines cut ...]
   linearizeOplusNs(std::make_index_sequence<_nr_of_vertices>());
                                                              ^
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp: In member function ‘virtual void g2o::BaseFixedSizedEdge<D, E, VertexTypes>::mapHessianMemory(number_t*, int, int, bool)’:
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:229:57: error: ‘_hessianTupleTransposed’ was not declared in this scope
     tuple_apply_i(MapHessianMemoryK{d, vj_dim, vi_dim}, _hessianTupleTransposed,
                                                         ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:229:57: note: suggested alternative: ‘HessianTupleType’
     tuple_apply_i(MapHessianMemoryK{d, vj_dim, vi_dim}, _hessianTupleTransposed,
                                                         ^~~~~~~~~~~~~~~~~~~~~~~
                                                         HessianTupleType
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:232:57: error: ‘_hessianTuple’ was not declared in this scope
     tuple_apply_i(MapHessianMemoryK{d, vi_dim, vj_dim}, _hessianTuple,
                                                         ^~~~~~~~~~~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:232:57: note: suggested alternative: ‘HessianTupleType’
     tuple_apply_i(MapHessianMemoryK{d, vi_dim, vj_dim}, _hessianTuple,
                                                         ^~~~~~~~~~~~~
                                                         HessianTupleType
modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/build.make:62: recipe for target 'modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/src/visual_slam_factory.cpp.o' failed
make[2]: *** [modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/src/visual_slam_factory.cpp.o] Error 1
CMakeFiles/Makefile2:404: recipe for target 'modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/all' failed
make[1]: *** [modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

`

RainerKuemmerle, 2021-03-07:
You need C++14 activated on your compiler.

Airplane5, 2021-03-07:
@RainerKuemmerle 
thank u for ur reply!
I add "set(CMAKE_CXX_STANDARD 14_)" into CMakeLists.txt, but problem still occured.
Should I do something else?
My g++ version is 7.5.0.

sjulier, 2021-03-07:
It should be set(CMAKE_CXX_STANDARD 14) - no underscore after 14. Does that make any difference?

Also are the compile arguments being anywhere being used to set the c++ standard? I'm not sure of the order of precedence.

jeychandar, 2022-12-21:
I have set(CMAKE_CXX_STANDARD 14) but still showing the same error. I am not sure what happened

closed 2025-01-31 by github-actions[bot]