RainerKuemmerle/g2o #502: Updating vertex estimates

MrGrayCode, 2021-04-03:
For a camera calibration graph I wish to perturb the initial pose estimates of the graph. Is there a way to do this other than removing the vertex and replacing it with a new one? Essentially I just need to update the estimate passed to a vertex using `setEstimate`

sjulier, 2021-04-03:
I'm a bit unclear about the question - why can't you just call estimate() on the vertices you want to change, perturb the value, and set the result using setEstimate()?

If you aren't sure of the vertex class, you should be able to set estimateDate() and setEstimateData() instead.

MrGrayCode, 2021-04-05:
Thank you for your response @sjulier. To make the question a bit more clear, when I access the vertices of the SparseOptimizer graph using `graph->vertices()` the vertices are of the class `HyperGraph::Vertex` so I cannot use `estimate` and `setEstimate`. I will try `setEstimateData` as you suggested.

RainerKuemmerle, 2021-04-05:
You need to cast them, to the correct class either using static_cast or dynamic_cast.
See
https://en.cppreference.com/w/cpp/language/dynamic_cast
https://en.cppreference.com/w/cpp/language/static_cast

Or you maintain a copy of the pointer during the lifetime of the optimizer to access and manipulate the estimate. See, for example,
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/examples/data_fitting/circle_fit.cpp

closed 2021-04-06 by MrGrayCode