RainerKuemmerle/g2o #503: g2o version for Ubuntu18.04

JackRen88, 2021-04-22:
Hello,I wonder that  to use which version of g2o in ubuntu 18.04.

pauljnixon, 2021-04-29:
I'm using Ubuntu 18.04 also and I was able to build from source using the current head of master (the default you get when cloning from github) no problem.  Use the instructions in the Readme.  

I haven't tried to build the GUI, so if that's what you're having trouble with, I can't help you.

43998213, 2021-09-30:
> I'm using Ubuntu 18.04 also and I was able to build from source using the current head of master (the default you get when cloning from github) no problem. Use the instructions in the Readme.
> 
> I haven't tried to build the GUI, so if that's what you're having trouble with, I can't help you.

install g2o on jetson nano got error:   c++: internal compiler error
I tried
sudo dd if=/dev/zero of=/swapfile bs=64M count=32
sudo mkswap /swapfile
chmod 0600 /swapfile  
sudo swapon /swapfile
I tried gcc7. gcc6. gcc5 
but all failed
help please

closed 2024-02-18 by RainerKuemmerle