RainerKuemmerle/g2o #504: Problem with calibrating SLAM: Segmentation fault for quadratic form of BaseFixedSizeEdge<3, SE2, VertexSE2, VertexSE2, VertexSE2>

artjvl, 2021-04-23:
Hi,

I am trying to implement a ternary hyper-edge, akin to `g2o/types/sclam2d/edge_se2_sensor_calib`, but I am having trouble optimizing a graph containing this edge.

For this edge `EdgePoses2DSE2PSE2 : public BaseFixedSizedEdge<3, SE2, VertexSE2, VertexSE2, VertexSE2>` with tag `CONSTRAINT_POSES2D_SE2_PSE2`, I have defined the `computeErrror` method as:
```
void computeError() {
    const VertexSE2* v1     = static_cast<const VertexSE2*>(_vertices[0]);
    const VertexSE2* v2     = static_cast<const VertexSE2*>(_vertices[1]);
    const VertexSE2* offset-= static_cast<const VertexSE2*>(_vertices[2]);
    const SE2& x1 = v1->estimate();
    const SE2& x2 = v2->estimate();
    SE2 delta = _inverseMeasurement * ((x1 * offset->estimate()).inverse() * x2 * offset->estimate());
    _error = delta.toVector();
}
```

I have generated the following graph, saved in a `.g2o` file, which simulates 10 steps forward and 10 steps backwards, with a loop-closure between the first and last pose:

```
NODE_SE2 1 0 0 0
FIX 1
NODE_SE2 2 0 0 0
NODE_SE2 3 0.9802 0.00128 0.07651
NODE_SE2 4 1.94214 0.04612 0.06036
NODE_SE2 5 2.89714 0.15208 -0.05234
NODE_SE2 6 3.83821 0.11379 -0.0258
NODE_SE2 7 4.84846 0.07349 -0.02184
NODE_SE2 8 5.90449 0.00628 -0.04111
NODE_SE2 9 6.97665 -0.05104 0.08347
NODE_SE2 10 7.93156 0.03124 0.08839
NODE_SE2 11 8.85674 0.10356 0.07637
NODE_SE2 12 9.78573 0.19927 -3.03362
NODE_SE2 13 8.81109 0.10037 -2.94964
NODE_SE2 14 7.85955 -0.0802 -2.90932
NODE_SE2 15 6.89482 -0.35877 -2.98672
NODE_SE2 16 5.87089 -0.51041 -3.01956
NODE_SE2 17 4.8941 -0.66012 -3.04686
NODE_SE2 18 3.86703 -0.81211 -3.11581
NODE_SE2 19 2.84514 -0.86475 -3.09083
NODE_SE2 20 1.8694 -0.91707 -3.06156
NODE_SE2 21 0.89101 -1.04016 -2.99387
NODE_SE2 22 -0.08696 -1.22328 0.14418
CONSTRAINT_POSES2D_SE2_PSE2 1 3 2 0.9802 0.00128 0.07651 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 3 4 2 0.96255 -0.02883 -0.01616 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 4 5 2 0.95966 0.04817 -0.11269 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 5 6 2 0.94178 0.01099 0.02653 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 6 7 2 1.01095 -0.01422 0.00396 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 7 8 2 1.05725 -0.04413 -0.01927 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 8 9 2 1.0736 -0.01321 0.12459 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 9 10 2 0.95845 0.00237 0.00492 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 10 11 2 0.92795 -0.00964 -0.01203 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 11 12 2 0.93359 0.02456 -3.10998 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 12 13 2 0.97962 -0.00671 0.08397 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 13 14 2 0.96851 -0.00428 0.04032 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 14 15 2 1.00296 0.04903 -0.0774 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 15 16 2 1.03506 -0.00813 -0.03285 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 16 17 2 0.98775 0.02971 -0.02729 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 17 18 2 1.03685 0.05414 -0.06896 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 18 19 2 1.0229 0.02628 0.02498 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 19 20 2 0.97714 0.00275 0.02927 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 20 21 2 0.98509 0.04448 0.06769 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 21 22 2 0.99427 0.03718 3.13805 800 0 0 600 0 400
CONSTRAINT_POSES2D_SE2_PSE2 1 22 2 0.00706 0.04984 -0.02723 800 0 0 600 0 400
```
Here, `NodeSE2` with id 2 represent the sensor offset.

Using the g2o cli application with any solver, a segmentation fault occurs when building the system, e.g. at `core/optimization_algorithm_gauss_newton.cpp:72: _solver.buildSystem();`. More specifically, the method `core/base_fixed_sized_edge.hpp:87: BaseFixedSizedEdge<D, E, VertexTypes...>::constructOffDiagonalQuadraticFormM()` seems to trigger the error in the Eigen library.

Is it possible to use edges of type `BaseFixedSizedEdge<3, SE2, VertexSE2, VertexSE2, VertexSE2>` in the way that I have intended? Could you give me some insight into how to fix this problem?

RainerKuemmerle, 2021-04-23:
Can you provide access to a minimal working example?

artjvl, 2021-04-23:
Hello Rainer,

Thanks for the quick response!

Please find below my minimal working example:
- I have attached the graph I used here: [graph.zip](https://github.com/RainerKuemmerle/g2o/files/6368240/graph.zip)
- I have attached the definition of my ternary edge `ConstraintPoses2DSE2PSE2 : public BaseFixedSizedEdge<3, SE2, VertexSE2, VertexSE2, VertexSE2>` to `g2o/g2o/types/sclam2d` (i.e., added files `constraint_poses2d_se2_pse2.h` and `constraint_poses2d_se2_pse2.cpp` and modified `CMakeLists.txt`, `types_sclam2d.h` and `types_sclam2d.cpp`). This folder is attached here: [sclam2d.zip](https://github.com/RainerKuemmerle/g2o/files/6368241/sclam2d.zip).
- For convenience, I have redundantly added the entire `g2o/g2o` subfolder here: [g2o.zip](https://github.com/RainerKuemmerle/g2o/files/6368239/g2o.zip).

(If there is a better way to share the minimal working example, please let me know!)

I have compiled the code as per your README, and ran the cli with:
```
./g2o -v graph.g2o
```
This triggers the segmentation fault, which I believe occurs when building the system, and specifically the construction of the quadratic form.

If you have any questions, please let me know.
Thanks again!

artjvl, 2021-04-30:
Hello @RainerKuemmerle,

Could you perhaps give me a pointer on where to look for a solution to this problem?
I'm still struggling to get my optimisation working for `BaseFixedSizeEdge`s with 3 (or more?) connecting nodes.

Hoping to get this resolved soon.
Thanks!

RainerKuemmerle, 2021-05-13:
Thanks for reporting this issue. I think I found the cause.
@artjvl can you please double check whether the above commit works for you?

After optimizing I obtain a sensor offset of  -0.0280781 -0.285836 0.0368882
Btw. I assume EdgeSE2SensorCalib would work in your case as well?

commit 5c30e3ca (2021-05-15): Store rowmajor flag for each Hessian block

The BaseFixedSizedEdge template needs to store for each Hessian block if
the respective block is transposed or not. Before a single boolean was
used which means that for an edge all blocks had to be on the upper
triangular block of the Hessian.

fixes #504
commit 80b6e3aa (2021-10-09): Store rowmajor flag for each Hessian block

The BaseFixedSizedEdge template needs to store for each Hessian block if
the respective block is transposed or not. Before a single boolean was
used which means that for an edge all blocks had to be on the upper
triangular block of the Hessian.

fixes #504
commit 97521b4e (2021-10-09): Store rowmajor flag for each Hessian block

The BaseFixedSizedEdge template needs to store for each Hessian block if
the respective block is transposed or not. Before a single boolean was
used which means that for an edge all blocks had to be on the upper
triangular block of the Hessian.

fixes #504
closed 2021-10-09 by RainerKuemmerle