RainerKuemmerle/g2o #508: Compile error on Ubuntu 20.04

SakuraWHU, 2021-05-10:
I just run following commands
```
git clone https://github.com/RainerKuemmerle/g2o.git
mkdir build
cd build
cmake ../
make
```

and errors occur
```
[  4%] Built target stuff
[ 16%] Built target core
[ 18%] Built target g2o_cli_library
[ 19%] Built target g2o_cli_application
[ 28%] Built target types_slam3d
[ 32%] Built target g2o_hierarchical_library
[ 33%] Built target g2o_hierarchical_application
[ 40%] Built target types_slam2d
[ 44%] Built target types_data
[ 53%] Built target types_sba
[ 54%] Built target types_sim3
[ 55%] Built target types_icp
[ 58%] Built target types_sclam2d
[ 63%] Built target types_slam2d_addons
[ 68%] Built target types_slam3d_addons
[ 69%] Built target solver_pcg
[ 70%] Built target solver_dense
[ 71%] Built target solver_eigen
[ 72%] Built target solver_slam2d_linear
[ 73%] Built target solver_structure_only
[ 74%] Built target ba_demo
[ 74%] Built target ba_anchored_inverse_depth_demo
[ 75%] Building CXX object g2o/examples/bal/CMakeFiles/bal_example.dir/bal_example.cpp.o
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:96:52: error: template argument 2 is invalid
   96 | class VertexPointBAL : public BaseVertex<3, Vector3> {
      |                                                    ^
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp: In member function ‘virtual void VertexPointBAL::oplusImpl(const double*)’:
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:114:5: error: reference to ‘Vector3’ is ambiguous
  114 |     Vector3::ConstMapType v(update);
      |     ^~~~~~~
In file included from /usr/local/include/eigen3/Eigen/Core:295,
                 from /home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:27:
/usr/local/include/eigen3/Eigen/src/Core/Matrix.h:524:7: note: candidates are: ‘template<class Type> using Vector3 = Eigen::Matrix<Type, 3, 1>’
  524 | using Vector##SizeSuffix = Matrix<Type, Size, 1>;                 \
      |       ^~~~~~
/usr/local/include/eigen3/Eigen/src/Core/Matrix.h:541:1: note: in expansion of macro ‘EIGEN_MAKE_TYPEDEFS’
  541 | EIGEN_MAKE_TYPEDEFS(3, 3)
      | ^~~~~~~~~~~~~~~~~~~
In file included from /home/xuqi/pkgs/g2o/g2o/core/auto_differentiation.h:34,
                 from /home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:32:
/home/xuqi/pkgs/g2o/g2o/core/eigen_types.h:50:9: note:                 ‘using Vector3 = g2o::VectorN<3>’
   50 |   using Vector3 = VectorN<3>;
      |         ^~~~~~~
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:115:5: error: ‘_estimate’ was not declared in this scope
  115 |     _estimate += v;
      |     ^~~~~~~~~
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:115:18: error: ‘v’ was not declared in this scope
  115 |     _estimate += v;
      |                  ^
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:113:40: warning: unused parameter ‘update’ [-Wunused-parameter]
  113 |   virtual void oplusImpl(const double* update) {
      |                          ~~~~~~~~~~~~~~^~~~~~
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp: At global scope:
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:141:93: error: template argument 2 is invalid
  141 | class EdgeObservationBAL : public BaseBinaryEdge<2, Vector2, VertexCameraBAL, VertexPointBAL> {
      |                                                                                             ^
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp: In member function ‘bool EdgeObservationBAL::operator()(const T*, const T*, T*) const’:
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:197:26: error: there are no arguments to ‘measurement’ that depend on a template parameter, so a declaration of ‘measurement’ must be available [-fpermissive]
  197 |     error = prediction - measurement().cast<T>();
[... 129 lines cut ...]
      |                      ^~~~~~~~
In file included from /home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:32:
/home/xuqi/pkgs/g2o/g2o/core/auto_differentiation.h: In instantiation of ‘static void g2o::AutoDifferentiation<Edge, EstimateAccess>::computeError(Edge*) [with Edge = EdgeObservationBAL; EstimateAccess = g2o::EstimateAccessor<EdgeObservationBAL>]’:
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:202:3:   required from here
/home/xuqi/pkgs/g2o/g2o/core/auto_differentiation.h:156:35: error: ‘Dimension’ is not a member of ‘EdgeObservationBAL’
  156 |     static_assert(Edge::Dimension > 0, "Dynamically sized edges are not supported");
      |                         ~~~~~~~~~~^~~
/home/xuqi/pkgs/g2o/g2o/core/auto_differentiation.h:157:31: error: ‘_nr_of_vertices’ is not a member of ‘EdgeObservationBAL’
  157 |     computeErrorNs(that, std::make_index_sequence<Edge::_nr_of_vertices>());
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xuqi/pkgs/g2o/g2o/core/auto_differentiation.h: In instantiation of ‘static void g2o::AutoDifferentiation<Edge, EstimateAccess>::linearize(Edge*) [with Edge = EdgeObservationBAL; EstimateAccess = g2o::EstimateAccessor<EdgeObservationBAL>]’:
/home/xuqi/pkgs/g2o/g2o/examples/bal/bal_example.cpp:202:3:   required from here
/home/xuqi/pkgs/g2o/g2o/core/auto_differentiation.h:168:35: error: ‘Dimension’ is not a member of ‘EdgeObservationBAL’
  168 |     static_assert(Edge::Dimension > 0, "Dynamically sized edges are not supported");
      |                         ~~~~~~~~~~^~~
/home/xuqi/pkgs/g2o/g2o/core/auto_differentiation.h:169:33: error: ‘_nr_of_vertices’ is not a member of ‘EdgeObservationBAL’
  169 |     linearizeOplusNs(that, std::make_index_sequence<Edge::_nr_of_vertices>());
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [g2o/examples/bal/CMakeFiles/bal_example.dir/build.make:63：g2o/examples/bal/CMakeFiles/bal_example.dir/bal_example.cpp.o] error 1
make[1]: *** [CMakeFiles/Makefile2:1652：g2o/examples/bal/CMakeFiles/bal_example.dir/all] error 2
make: *** [Makefile:152：all] error 2
```

I am so confused !
does anyone can help me?

RainerKuemmerle, 2021-05-13:
Thanks for reporting the issue. Can you see whether above commit works on your setup.
I removed using Eigen's namespace from the example which should avoid the ambiguous type Vector3.

rshivansh, 2021-05-13:
@RainerKuemmerle I was facing the same issue earlier to the latest commit you made, I still face the same issue. Can you please help with this? Thanks

rshivansh, 2021-05-15:
@SakuraWHU were you able to solve the issue? Thanks

RainerKuemmerle, 2021-05-15:
Now also dropping g2o namespace in this example. Note that above commits are on a branch.
Does this change your compiler output? Please note that compiling on Travis running 20.04 was fine.

SakuraWHU, 2021-05-15:
> Now also dropping g2o namespace in this example. Note that above commits are on a branch.
> Does this change your compiler output? Please note that compiling on Travis running 20.04 was fine.

Thank you! It really solves my problem.
now, I can run g2o on Ubuntu 20.04!

PR #512 (2021-05-15): Rainer/issues508
Closes #508
commit 954b271c (2021-05-15): Rainer/issues508 (#512)

* Drop using g2o and Eigen's namespace in bal example

Fix #508
commit 5d70eb1d (2021-05-15): Rainer/issues508 (#512)

* Drop using g2o and Eigen's namespace in bal example

Fix #508
closed 2021-05-15 by RainerKuemmerle