RainerKuemmerle/g2o #510: Compilation error on Ubuntu 16.04 with Eigen3.3.9

songlin, 2021-05-13:
```
[ 70%] Building CXX object g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/solver_eigen.cpp.o
In file included from /home/songlin/Projects/g2o/g2o/solvers/eigen/solver_eigen.cpp:35:0:
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h: In instantiation of ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’:
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:114:44:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::computeCholesky(const g2o::SparseBlockMatrix<MatrixType>&, double&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:89:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, number_t*, number_t*) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>; number_t = double]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/solver_eigen.cpp:107:1:   required from here
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:150:90: error: no type named ‘StorageIndex’ in ‘class g2o::LinearSolverEigen<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::CholeskyDecomposition’
         using Ordering = Eigen::AMDOrdering<typename CholeskyDecomposition::StorageIndex>;
                                                                                          ^
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h: In instantiation of ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 6, 6, 0, 6, 6>]’:
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:114:44:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::computeCholesky(const g2o::SparseBlockMatrix<MatrixType>&, double&) [with MatrixType = Eigen::Matrix<double, 6, 6, 0, 6, 6>]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:89:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, number_t*, number_t*) [with MatrixType = Eigen::Matrix<double, 6, 6, 0, 6, 6>; number_t = double]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/solver_eigen.cpp:107:1:   required from here
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:150:90: error: no type named ‘StorageIndex’ in ‘class g2o::LinearSolverEigen<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::CholeskyDecomposition’
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h: In instantiation of ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 3, 3>]’:
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:114:44:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::computeCholesky(const g2o::SparseBlockMatrix<MatrixType>&, double&) [with MatrixType = Eigen::Matrix<double, 3, 3>]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:89:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, number_t*, number_t*) [with MatrixType = Eigen::Matrix<double, 3, 3>; number_t = double]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/solver_eigen.cpp:107:1:   required from here
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:150:90: error: no type named ‘StorageIndex’ in ‘class g2o::LinearSolverEigen<Eigen::Matrix<double, 3, 3> >::CholeskyDecomposition’
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h: In instantiation of ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, -1, -1>]’:
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:114:44:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::computeCholesky(const g2o::SparseBlockMatrix<MatrixType>&, double&) [with MatrixType = Eigen::Matrix<double, -1, -1>]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:89:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, number_t*, number_t*) [with MatrixType = Eigen::Matrix<double, -1, -1>; number_t = double]’
/home/songlin/Projects/g2o/g2o/solvers/eigen/solver_eigen.cpp:107:1:   required from here
/home/songlin/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:150:90: error: no type named ‘StorageIndex’ in ‘class g2o::LinearSolverEigen<Eigen::Matrix<double, -1, -1> >::CholeskyDecomposition’
g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/build.make:62: recipe for target 'g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/solver_eigen.cpp.o' failed
make[2]: *** [g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/solver_eigen.cpp.o] Error 1
CMakeFiles/Makefile2:1808: recipe for target 'g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/all' failed
make[1]: *** [g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

Hey there, has anyone encoutered above compilation problem?
I am compiling at ubuntu 16.04 with Eigen 3.3.9, I googled a little bit but no luck.
Thanks

RainerKuemmerle, 2021-05-13:
Does the same happen with Eigen 3.3.7?
Locally my setup is Ubuntu 20.10 with Eigen 3.3.7. Travis on 20.04 is also with Eigen 3.3.7.
Windows with Eigen 3.3.9 seems fine though.

Maybe the compiler shipped within 16.04 is not suited with most recent versions of either g2o or Eigen.

songlin, 2021-05-13:
Thank you for the reply. 
I didnot try Eigen 3.3.7.  But after i change the line to:
```
using Ordering = Eigen::AMDOrdering<typename SparseMatrix::StorageIndex>;
```
it compiled successfully, but another problem happened [issue 424.](https://github.com/RainerKuemmerle/g2o/issues/424)   
Then i realized the real problem is that cmake found the wrong version of Eigen in my computer.

Then i specify the right version this time:
```
cmake ../ -DEIGEN3_INCLUDE_DIR=/usr/local/include/eigen3
```

Problem solved. Hope this can help someone.

By the same way, i see a lot of examples in the source folder `g2o/examples `, but i couldn't find any instructions about how to run them. Thanks again.

closed 2021-05-13 by songlin

Chrislzy1993, 2021-12-08:
@songlin hello, i meet same problem too, and i use eigen3.3.9, when i use eigen3.2.9, there is no problem

songlin, 2021-12-12:
@Chrislzy1993 thanks, good to know.