RainerKuemmerle/g2o #511: The meaning of the BlockSolverTraits<a,b>

zhaozhongch, 2021-05-14:
When we define the block solver, we need to specify two numbers. For example, in orbslam2, g2o/block_solver.h, it has
```
typedef BlockSolver< BlockSolverTraits<6, 3> > BlockSolver_6_3;  
```
And in optimizer.cc local bundle adjustment function it has
```
g2o::BlockSolver_6_3::LinearSolverType * linearSolver;
```
Previously, I think the `6` is the pose dimension, which is the variable dimension we want to optimize. `3` is the error (which is the edge) dimension. However, for a binary edge, we want to optimize 6DOF pose + 3DOF point = 9 variable dimension. So previously I think it should be `BlockSolver_9_3`. This looks wrong.
Then, some people told me 6 and 3 are both vertex dimensions. This makes sense for a binary edge that has two vertexes. But finally, I feel it is also wrong. Because in the `PoseOptimization`, we only have a unary edge, which means it only has 1 vertex. But in that function, it still uses `BlockSolver_6_3` to define the solver. Only 6 is meaningful because it stands for the 6 DOF pose. Then 3 should not mean `another vertex` because it only has 1 vertex.

How could I set the `a,b` in `BlockSolverTraits<a,b>`???? What exactly does it mean??

RainerKuemmerle, 2021-05-16:
Within `BlockSolverTraits<a,b>` `a` and `b` refer to the dimension of the vertices that are part of certain blocks when computing the Schur complement.
See Section III.D of the original paper: http://ais.informatik.uni-freiburg.de/publications/papers/kuemmerle11icra.pdf
In (24) `a` and `b` are denoted by `p` and `l`, respectively.

If you only have poses of dimension `a` you can use any `BlockSolverTraits<a,X>` regardless of how the edges are structured.
Specifying a fixed size here helps to generate compile time code for the matrix multiplications. If this efficiency gain is not important in your application, use `BlockSolverTraits<Eigen::Dynamic, Eigen::Dynamic>` instead.

zhaozhongch, 2021-05-16:
Thanks for the clarification! One more question is, what if I use `BaseMultiEdge`? For an edge that has multiple vertexes, how could I use `BlockSolverTraits<a,b>` since it only defines two vertexes dimensions?

macTracyHuang, 2022-10-17:
> Thanks for the clarification! One more question is, what if I use `BaseMultiEdge`? For an edge that has multiple vertexes, how could I use `BlockSolverTraits<a,b>` since it only defines two vertexes dimensions?

if it's for part of certain blocks when computing the Schur complement. I think maybe you need to set it as maximum dimension of the vertices between the edges?

closed 2025-01-31 by github-actions[bot]