RainerKuemmerle/g2o #516: How to use float instead of double?

Zero-Swangel, 2021-05-21:
I found an option G2O_SINGLE_PRECISION_MATH in `config.h`, but after defined it I got an error
```
/usr/bin/ld: /home/ros_ws/src/track_lidar_back_end/lib/libtrack_lidar_back_end.so: undefined reference to `g2o::MarginalCovarianceCholesky::computeCovariance(float**, std::vector<int, std::allocator<int> > const&)'
/usr/bin/ld: /home/ros_ws/src/track_lidar_back_end/lib/libtrack_lidar_back_end.so: undefined reference to `g2o::MarginalCovarianceCholesky::setCholeskyFactor(int, int*, int*, float*, int*)'
/usr/bin/ld: /home/ros_ws/src/track_lidar_back_end/lib/libtrack_lidar_back_end.so: undefined reference to `g2o::MarginalCovarianceCholesky::computeCovariance(g2o::SparseBlockMatrix<Eigen::Matrix<float, -1, -1, 0, -1, -1> >&, std::vector<int, std::allocator<int> > const&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&)'
```
I found this function in `/usr/local/include/g2o/core/marginal_covariance_cholesky.h`, but it is declared in number_t which is defined as float.

sjulier-overleaf, 2021-06-26:
Is this build time or run time?

etherswangel, 2023-09-08:
Sorry I forgot to reply... it's build time. Closing as I don't use g2o now, sry

closed 2023-09-08 by etherswangel