RainerKuemmerle/g2o #520: How to use unique_ptr for g2o

deeprun, 2021-06-08:
Hi, I used to use the code below:

`
typedef` g2o::BlockSolver< g2o::BlockSolverTraits<6, 3> > Block;
Block::LinearSolverType* linearSolver = new g2o::LinearSolverCSparse<Block::PoseMatrixType>(); 
Block* solver_ptr = new Block(linearSolver); 
g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(solver_ptr);
`

But this code does not compile with the new g2o code. How can I fix it?

sjulier-overleaf, 2021-06-26:
Please check the examples in the repository. Some of them use factories to construct types, and others work directly without the use of the factories.

whu-lyh, 2021-07-27:
checkout to a old version g2o, and this works.

closed 2021-10-09 by RainerKuemmerle