RainerKuemmerle/g2o #521: /usr/bin/ld: ../../../../lib/libg2o_solver_cholmod.so: undefined reference to `SuiteSparse_time'

zhaoworking, 2021-06-09:
$ make
[  1%] Built target freeglut_minimal
[  2%] Built target opengl_helper
[  5%] Built target stuff
[ 14%] Built target core
[ 15%] Built target g2o_cli_library
[ 16%] Built target g2o_cli_application
[ 23%] Built target types_slam3d
[ 25%] Built target g2o_hierarchical_library
[ 26%] Built target g2o_hierarchical_application
[ 30%] Built target types_slam3d_addons
[ 36%] Built target types_slam2d
[ 39%] Built target types_slam2d_addons
[ 45%] Built target g2o_simulator_library
[ 45%] Built target convertSegmentLine_application
[ 45%] Built target g2o_simulator2d_application
[ 46%] Built target g2o_anonymize_observations_application
[ 47%] Built target g2o_simulator3d_application
[ 52%] Built target viewer_library
[ 53%] Built target g2o_viewer
[ 56%] Built target types_data
[ 63%] Built target types_sba
[ 63%] Built target types_sim3
[ 64%] Built target types_icp
[ 66%] Built target types_sclam2d
[ 67%] Built target solver_pcg
[ 67%] Built target solver_dense
[ 68%] Built target solver_eigen
[ 69%] Built target solver_slam2d_linear
[ 69%] Built target solver_structure_only
[ 70%] Built target csparse_extension
[ 71%] Built target solver_csparse
[ 72%] Built target solver_cholmod
[ 72%] Linking CXX executable ../../../../bin/ba_demo
/usr/bin/ld: ../../../../lib/libg2o_solver_cholmod.so: undefined reference to `SuiteSparse_time'
collect2: error: ld returned 1 exit status
g2o/examples/ba/CMakeFiles/ba_demo.dir/build.make:111: recipe for target '../bin/ba_demo' failed
make[2]: *** [../bin/ba_demo] Error 1
CMakeFiles/Makefile2:2042: recipe for target 'g2o/examples/ba/CMakeFiles/ba_demo.dir/all' failed
make[1]: *** [g2o/examples/ba/CMakeFiles/ba_demo.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
I have installed the SuiteSparse library on my Ubuntu 18.04, but it occures to the problem of SuiteSparse_time error. I see a well-working merge with the problem on 2013, why it still exits in my OS?

sjulier-overleaf, 2021-06-26:
I've never seen this error myself, and I've built on both 18.04 and 20.04. Which version of SuiteSparse are you using and how as it built? I've seen some posts from 2013-2015 which suggest some flags weren't defined, but this error hasn't been reported in the past 2-3 years.

huziqi, 2021-12-27:
try this commit, it fix my problem
e141a1c2d2e2f4277b551b6aec8f7c7b67008e40

sandeepnmenon, 2022-05-09:
+1
I am encountering same error. I installed Suitsparse by downloading from the source and installed using Cmake. 
Also tried installing `libsuitesparse-dev`, but still same error

Environment details
Ubuntu 20.04
SuiteSparse version: 4.2.1

sandeepnmenon, 2022-05-09:
I had to switch off `G2O_BUILD_EXAMPLES` flag for a successfull make
Ran
`cmake -DG2O_BUILD_EXAMPLES=OFF ../`
Then used make install to install the libraries. Hope this is fine.

closed 2025-01-30 by github-actions[bot]