RainerKuemmerle/g2o #528: Segmentation fault optimizing with EdgeSE3PointXYZ

polifab, 2021-07-05:
Hello everyone. 

I receive the following message from gdb when I try to call  g2o::SparseOptimizer::optimize () in a graph that has 2 vertex SE3, which one is fixed, and one vertex PointXYZ ( representing Landmark ), 2 EdgeSE3 connecting the 2 vertex SE3 and a EdgeSE3PointXYZ connecting a vertex SE3 to a vertex PointXYZ. The solver is "lm_var_cholmod"

```
0x00007ffff7c3d042 in double __vector(2) Eigen::internal::pload<double __vector(2)>
(Eigen::internal::unpacket_traits<double__vector(2)>::type const*) () from /usr/local/lib/libg2o_core.so

```

If I use the solver "lm_fix6_3_cholmod" it throws another error:

```
graph_optimization_node: /usr/include/Eigen/src/Core/PlainObjectBase.h:281: void
Eigen::PlainObjectBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = 
Eigen::Matrix<double, 6, 6, 0>; Eigen::Index = long int]: Assertion `(!(RowsAtCompileTime!=Dynamic) ||
(rows==RowsAtCompileTime)) && (!(ColsAtCompileTime!=Dynamic) || (cols==ColsAtCompileTime)) &&
(!(RowsAtCompileTime==Dynamic && MaxRowsAtCompileTime!=Dynamic) || (rows<=MaxRowsAtCompileTime)) &&
(!(ColsAtCompileTime==Dynamic && MaxColsAtCompileTime!=Dynamic) || (cols<=MaxColsAtCompileTime)) && rows>=0 &&
cols>=0 && "Invalid sizes when resizing a matrix or array."' failed.
```

It works fine with only VertexSE3 and EdgeSE3. It is launched in Ubuntu 20.04, with the last g2o version and Eigen versrion 3.3.7. Has anyone any idea how to solve this? Maybe with some compilation flag.

Thank you in advance for your attention.

sjulier, 2021-07-09:
Can you post the code for an example?

polifab, 2021-07-10:
Yes! So, this is how I initialize the SparseOptimizer:

```cpp
    g2o::OptimizationAlgorithmProperty solverProperty;
    std::string solverName = "lm_var_cholmod";
    optimizer->setVerbose(true);
    optimizer->setAlgorithm(
        g2o::OptimizationAlgorithmFactory::instance()->construct(solverName, solverProperty));
  // add the parameter representing the sensor offset
    g2o::Isometry3 sensorOffsetTransf;
    sensorOffsetTransf.setIdentity();
    g2o::ParameterSE3Offset* sensorOffset = new g2o::ParameterSE3Offset;
    sensorOffset->setOffset(sensorOffsetTransf);
    sensorOffset->setId(0);
    optimizer->addParameter(sensorOffset);
    // initialize the graph with the map node
    Eigen::Vector3d pos(0,0,0);
    Eigen::Quaterniond q;
    q.setIdentity();
    if(!addPoseVertex(optimizer, 0, q, pos, true)){
        return false;
    }
```
with the utilities function *addPoseVertex*,:

```cpp
bool addPoseVertex(g2o::SparseOptimizer * optimizer, uint32_t vertex_id, Eigen::Quaterniond orientation, Eigen::Vector3d position, bool fixed)
{

// ------------------------------------------------ //
// Function addPoseVertex: It inserts a new REU pose node inside the graph optimizer
//      Input:
//              - g2o::SparseOptimizer * optimizer: the istance of the graph optimizer
//              - uint32_t vertex_id: node id (TODO choose convention)
//              - Eigen::Quaterniond orientation: quaternion containing orientation wrt map
//              - Eigen::Vector3d position: 3d vector containing REU position wrt map
//              - bool fixed: a boolean to indicate if the node is fixed
//      Output:
//              - a boolean set to true if the allocation of node went fine, false otherwise
// ------------------------------------------------ //

    g2o::SE3Quat pose(orientation, position);
    g2o::VertexSE3 * v_se3 = new g2o::VertexSE3();
    v_se3->setId(vertex_id);
    if(fixed){
        v_se3->setFixed(true);
    }
    v_se3->setEstimate(pose);

    if(optimizer->addVertex(v_se3)){
        return true;
    } else {
        return false;
    }
}
```

Then I add edgesSE3, landmark vertices and edgesSE3PointXYZ through:

```cpp
bool newPoseEdge(g2o::SparseOptimizer * optimizer, uint32_t id_0, uint32_t id_1, g2o::SE3Quat measurement, Eigen::Matrix<double, 6, 6>  information)
{
// ------------------------------------------------ //
// Function newPoseEdge: It inserts a new edge between 2 robots pose node inside the graph optimizer
//      Input:
//              - g2o::SparseOptimizer * optimizer: the istance of the graph optimizer
//              - uint32_t id_0: departing node id
//              - uint32_t id_1: arriving node id
//              - g2o::SE3Quat measurement: a g2o::SE3Quat containing the relative position and orientation of arriving node wrt departing node
//              - Eigen::Matrix<double, 6, 6>  information: covariance matrix of measurement
[... 74 lines cut ...]
        return false;
    }
}
```

At the moment I insert manually three nodes, one fixed in (0,0,0), one pose node VertexSE3 in (5,5,0) with orientation in quaternion = (0,0,0,1) and finally one landmark pose VertexPointXYZ in (10,10,0). I insert also one edgeSE3 between the fixed node and pose node and one edgeSE3PointXYZ. Then I call the optimization like this:

```cpp
      optimizer.initializeOptimization();
      std::cout << "Pre opt" << std::endl;
      auto start = high_resolution_clock::now();
      optimizer.optimize(10);
```

I also save the graph in the .g2o format:
```
PARAMS_SE3OFFSET 0 0 0 0 0 0 0 1 
VERTEX_SE3:QUAT 0 0 0 0 0 0 0 1 
FIX 0
VERTEX_SE3:QUAT 10000000 5 5 0 0 0 0 1 
VERTEX_TRACKXYZ 11000000 10 10 0 
EDGE_SE3:QUAT 0 10000000 5 5 0 0 0 0 1 1 0 0 0 0 0 1 0 0 0 0 1 0 0 0 1 0 0 1 0 1 
EDGE_SE3_TRACKXYZ 10000000 11000000 5 5 0 1 0 0 1 0 1 
```
Whatever solver I choose it gives me segmentation fault. I don't know if I'm missing something.

sjulier, 2021-07-14:
Thanks.

Hmm... I don't see anything obviously wrong with the code.

Would it be possible to post a self-contained example which builds but exhibits the crash?

RainerKuemmerle, 2021-07-17:
Regarding compiler flags. Please make sure that you have the same flags, in particular "-march".
You may try to put a minimal self-contained example in "g2o/playground" and some CMakeLists in there.
The main CMake of g2o will pick up the example.

polifab, 2021-08-12:
I was using release *20201223_git*.

Now using the last commit in master the problem seems disappeared.

I'm closing the issue, thank you.

closed 2021-08-12 by polifab

yanliang-wang, 2021-10-15:
> I was using release _20201223_git_.
> 
> Now using the last commit in master the problem seems disappeared.
> 
> I'm closing the issue, thank you.

Hi, I have the same issue as you. Can you tell me your g2o version?

polifab, 2021-10-15:
Hi, now I'm using master branch with commit:
```
commit a0585942a9e1a7255aa7c43452fd5fb2a8b869a4 (HEAD -> master)
Author: Simon Julier <s.julier@ucl.ac.uk>
Date:   Wed May 19 19:05:49 2021 +0100

    Conditional build of builtin types (#514)
    
    * Added cmake files to conditionally compile the different types, and compile the examples available with the different types.
    * Removed unneeded dependency.
    * Updated cmake files to support conditional compiling of types.

```

yanliang-wang, 2021-10-15:
Thank you for your reply. I tried your version but my code still doesn't work. I will try other solutions. Thank you again.

ayushsharma-crypto, 2022-02-17:
Can someone kindly suggest a suitable fix, if one exists, as I'm having the same problem? Even using the most recent commits does not significantly alter behavior. @RainerKuemmerle @polifab

RainerKuemmerle, 2022-02-17:
Minimal working example that demonstrates the problem would help a lot.

ayushsharma-crypto, 2022-02-17:
[noise.txt](https://github.com/RainerKuemmerle/g2o/files/8088028/noise.txt)

The above noise file is having a few robot noisy poses (SE3) and some landmarks (3D). In addition, odometry constraint, loop closure and observation edge. I believe the vertex and edge format to be correct(please confirm the same). So, when I try to run the following command from the ubuntu 20.04 terminal:
```
g2o -v -robustKernel Cauchy -robustKernelWidth 1 -o "opt.txt" -i 50 "noise.txt"
```

I get the following output:-
```
# Used Compiler: GNU /usr/bin/c++
# Using EigenSparseCholesky poseDim -1 landMarkDim -1 blockordering 1
Read input from noise.txt
Segmentation fault (core dumped)
```

But when I remove all `EDGE_SE3_TRACKXYZ` observation edges. It is running successfully without any issue/error.

RainerKuemmerle, 2022-02-17:
@ayushsharma-crypto your file does not contain any g2o::ParameterSE3Offset which is required for EdgeSE3PointXYZ to work. See code in this issue on how to associate it to the edge or for example also this test here
https://github.com/RainerKuemmerle/g2o/blob/d68c441622983a23d3761dca5d5f8f3074c717f1/unit_test/slam3d/jacobians_slam3d.cpp#L101:L136

Improvements for better handling of missing parameters are appreciated. At the moment, the code does not spot this but crashes.

Please create a new issue if the problem persists. But it does not seem related to the issue discussed in the original posting of this issue.