RainerKuemmerle/g2o #529: Connecting Vertex to more Optimizers

cirpote, 2021-07-09:
Hello everyone, 

I am trying to run a cascade of optimizations on the same vertexes, and i get the following error:

addVertex: FATAL, vertex with ID X has already registered with another graph Y

From what i understood, for each vertex, there is an internal OptimizableGraph pointer that must be NULL. In other words the same vertex cannot be associated to two different optimizers. 

My question is: Is there a manner to solve this by "un-assigning" vertexes to an optimizer?

Thanks anyway

sjulier, 2021-07-09:
By default, g2o takes ownership of the vertices and edges. Therefore, if you call removeVertex, the vertex will be deleted.

I think to stop this behaviour you have to build g2o with `G2O_NO_IMPLICIT_OWNERSHIP_OF_OBJECTS` set to `NO`. It won't automatically delete the vertices and edges anymore, so you could move them between the graphs.

I guess one question is whether you need to migrate the vertices between graphs. For example, could you use `setLevel(.)` to enable and disable different clusters of edges?

RainerKuemmerle, 2021-07-17:
In addition to the ownership, the internal book-keeping of how vertices/edges are connected would get confused.
Typically, one can run multiple optimizations on same vertices re-using the optimizer. For example, by selecting an appropriate subset of vertices/edges to construct the optimization problem.

closed 2025-01-30 by github-actions[bot]