RainerKuemmerle/g2o #533: base_fixed_sized_edge.hpp:165 has no member named ‘fill’

innistard, 2021-07-19:
I can make and make install g2o library, but when I use it in my code, I met an error when build my code:

/usr/local/include/g2o/core/base_variable_sized_edge.hpp: In member function ‘virtual void g2o::BaseVariableSizedEdge<D, E>::linearizeOplus()’:
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:84:12: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘fill’
     add_vi.fill(0.);
            ^
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:88:24: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘data’
       vi->oplus(add_vi.data());
                        ^
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:94:24: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘data’
       vi->oplus(add_vi.data());

mtlazaro, 2021-07-20:
Hi,
I was having the same problem, my code not compiling because the same error.

I had installed version 1.13.0 from ceres, the one indicated in [Cartographer instructions](https://google-cartographer.readthedocs.io/en/latest/)
I have updated ceres to the last version and my code has compiled again. (ubuntu 18 + last g2o version + last ceres version)

innistard, 2021-07-21:
> Hi,
> I was having the same problem, my code not compiling because the same error.
>
> I had installed version 1.13.0 from ceres, the one indicated in [Cartographer [...]

closed 2021-07-21 by innistard

504115681, 2021-11-01:
> Hi,
> I was having the same problem, my code not compiling because the same error.
> I had installed version 1.13.0 from ceres, the one indicated in [Cartographer [...]

Hello, have you solved it? I would appreciate it if you could help me!

mtlazaro, 2021-11-03:
Hi,
yes I solved the issue by updating the version of Ceres. Now I moved to
Ubuntu 20.04. In case it is more useful to you I detail here the last
configuration working for me:
Ubuntu 20.04:
- g2o tested until commit 5b14535b8d63f837699d1dde27fd350fc6738b2a
- ceres tested until commit 686428f5cddb8534fb63e4fcde014c0926ad4639