RainerKuemmerle/g2o #535: Can not compile custom edges after Merge request 506

johannes-graeter, 2021-07-20:
Thx for your great work!
After merge of 
https://github.com/RainerKuemmerle/g2o/pull/506
I can not compile my custom edges anymore (Setting G2O_USE_VENDORED_CERES of course).
Error:
/usr/local/include/g2o/core/base_fixed_sized_edge.h:37:12: fatal error: ceres/internal/fixed_array.h: No such file or directory
   37 |   #include <ceres/internal/fixed_array.h>

You help is much appreciated!

RainerKuemmerle, 2021-08-08:
Does https://github.com/RainerKuemmerle/g2o/pull/536 fix the problem for you?

MalcolmMielle, 2022-06-06:
I have the same problem with my custom edges on fedora 36. The PR sadly didn't fix it. Did you find a solution?

artofnothingness, 2022-06-12:
Installing libceres-dev on ubuntu solves the issue for me. So, probably you should install ceres libs

lzyplayer, 2023-01-11:
> Thx for your great work! After merge of #506 I can not compile my custom edges anymore (Setting G2O_USE_VENDORED_CERES of course). Error: /usr/local/include/g2o/core/base_fixed_sized_edge.h:37:12: [...]

Acutlly if you set `G2O_USE_VENDORED_CERES`  correctly, that line is supposed to be ignored 
g2o/core/base_fixed_sized_edge.h:37
```c++
#ifdef G2O_USE_VENDORED_CERES
#include "g2o/EXTERNAL/ceres/fixed_array.h"
#else
#include <ceres/internal/fixed_array.h>
#endif
```
It will search for `"g2o/EXTERNAL/ceres/fixed_array.h"` instead.

closed 2025-01-30 by github-actions[bot]