RainerKuemmerle/g2o #553: Mac OSX Catalina 10.15.7: brew install fails

lindzey, 2021-11-09:
I am attempting to install g2o following the instructions in the README (`brew install brewsci/science/g2o`), but it fails with:

~~~
==> Downloading https://archive.org/download/brewsci/bottles-science/g2o-2017-07-30.sierra.bottle.tar.gz
==> Downloading from https://ia601805.us.archive.org/0/items/brewsci/bottles-science/g2o-2017-07-30.sierra
#=#=#                                                                         
curl: (22) The requested URL returned error: 404 
Error: g2o: Failed to download resource "g2o"
Download failed: https://archive.org/download/brewsci/bottles-science/g2o-2017-07-30.sierra.bottle.tar.gz
~~~

HenryScratch, 2021-11-15:
Same on Big Sur. Any recommendations?

RainerKuemmerle, 2021-11-21:
I am not on MacOS. From the error it seems like the used formular uses an URL that does not exists.
Can it be fixed by using a release URL from github instead?
@sjulier Maybe you know more or can support?

sjulier, 2021-11-21:
I assume this is packaging on homebrew. I never set it up and I've never used it. I've only ever built from source.

I recall somebody set it up a couple of years ago, but I don't recall who. I think it's in the resolved issue list.

If not, would anybody be willing to volunteer to set it up?

closed 2025-01-29 by github-actions[bot]