RainerKuemmerle/g2o #571: Segmentation fault for SE3 case optimisation using g2o CLI

ayushsharma-crypto, 2022-02-18:
[noise_old.txt](https://github.com/RainerKuemmerle/g2o/files/8098772/noise_old.txt)

The above noise file is having a few robot noisy poses (SE3) and some landmarks (3D). In addition, odometry constraint, loop closure, and observation edge. I believe the vertex and edge format to be correct(please confirm the same). So, when I try to run the following command from the ubuntu 20.04 terminal:

```
g2o -v -robustKernel Cauchy -robustKernelWidth 1 -o "opt.txt" -i 50 "noise.txt"
```
I get the following output:-

```
# Used Compiler: GNU /usr/bin/c++
# Using EigenSparseCholesky poseDim -1 landMarkDim -1 blockordering 1
Read input from noise.txt
Segmentation fault (core dumped)
```
But when I remove all EDGE_SE3_TRACKXYZ observation edges. It is running successfully without any issue/error.
<hr>
This issue has been created because of a discussion on issue #528. @RainerKuemmerle  asked to put PARAMS_SE3OFFSET in the file for optimizing using g2o CLI, as  EdgeSE3PointXYZ won't work without it. However, the issue still remains the same. I don't know if I'm giving the wrong syntax for PARAMS_SE3OFFSET. But I used the following format:

```
PARAMS_SE3OFFSET id x y z qx qy qz qw
```
where `x y z qx qy qz qw` are having the same value as the corresponding robot pose vertex has for the given edge between robot pose & landmark. Below is the updated noise file.

[noise.txt](https://github.com/RainerKuemmerle/g2o/files/8098928/noise.txt)

However, this file is giving same output as above file.

Also, can someone please mention the correct format for giving g2o::ParameterSE3Offset to the noise file (i.e. what are the values to be written in front of PARAMS_SE3OFFSET in the noise file and what they denote) that I shared in the above chat, as I'm using g2o cli? I couldn't find it on the g2o wiki page.

RainerKuemmerle, 2022-02-18:
You need to add one Parameter to the edge. Not link a parameter as a vertex. The parameter describes an offset to apply on the pose vertex of the edge.

So on your file add as first line
`PARAMS_SE3OFFSET 0 0 0 0 0 0 0 1`

Add it to the EDGE_SE3_TRACKXYZ:
Change from
`EDGE_SE3_TRACKXYZ id1 id2 ....`
to
`EDGE_SE3_TRACKXYZ id1 id2 0`
where 0 is the param id

Using the noise.txt from #528 

We can do the above mentioned modifications on the file as follows
```
wget https://github.com/RainerKuemmerle/g2o/files/8088028/noise.txt
sed -i '1s/^/PARAMS_SE3OFFSET 0 0 0 0 0 0 0 1\n/' noise.txt
sed -i 's/^EDGE_SE3_TRACKXYZ \([0-9]\+\) \([0-9]\+\)/EDGE_SE3_TRACKXYZ \1 \2 0/' noise.txt
```

[image]

[noise.txt](https://github.com/RainerKuemmerle/g2o/files/8099567/noise.txt)

ayushsharma-crypto, 2022-02-21:
@RainerKuemmerle thank you for the reply. Now, the above command doesn't throw an error on execution. However, I would like to know the significance of this PARAMS_SE3OFFSET variable. Can you please mention the same?
Thanks in advance.

RainerKuemmerle, 2022-02-21:
It is not a variable of the optimization problem but a fixed offset between the pose of the vertex and the camera observing the points.
It allows to directly add measurements without projecting measurement along a kinematic chain.

ayushsharma-crypto, 2022-02-21:
@RainerKuemmerle  Okay, I get what you are saying. But, is there some kind of documentation or reference available? I would like to know more about it.

closed 2025-01-28 by github-actions[bot]