RainerKuemmerle/g2o #573: How I can compile g2o for Wayland (no X11)

pddl, 2022-02-20:
How to can compile g2o on Linux for Wayland i.e. NOX (no X11)?
Have tried some patching. It looks like too many thing in g2o rely on X11.
Appreciate if you could send a recipe for conda build for g2o for Wayland/Linux.

RainerKuemmerle, 2022-02-22:
I have not tried this myself. Where does it fail? Can you please give an example?

pddl, 2022-02-23:
Here is cmake option:
```
cmake ${CMAKE_ARGS} \
    [...]
    -DG2O_USE_OPENGL:BOOL=TRUE \
    -DOPENGL_USE_EGL=ON \
    ..
The issue is with OPENGL_gl_LIBRARY which does not exist.
```
find . -type f -exec grep -niH -e OPENGL_gl_LIBRARY {} ;
    ./g2o/types/slam2d_addons/CMakeLists.txt:21:  target_link_libraries(types_slam2d_addons ${OPENGL_gl_LIBRARY})
    ./g2o/types/slam3d_addons/CMakeLists.txt:26:  target_link_libraries(types_slam3d_addons ${OPENGL_gl_LIBRARY} )
    ./g2o/types/slam3d/CMakeLists.txt:49:  target_link_libraries(types_slam3d opengl_helper ${OPENGL_gl_LIBRARY} )
    ./g2o/types/sclam2d/CMakeLists.txt:16:  target_link_libraries(types_sclam2d ${OPENGL_gl_LIBRARY})
    ./g2o/types/slam2d/CMakeLists.txt:26:  target_link_libraries(types_slam2d opengl_helper ${OPENGL_gl_LIBRARY} )
    ./g2o/examples/slam2d/CMakeLists.txt:42:    ${OPENGL_gl_LIBRARY} ${OPENGL_glu_LIBRARY}
```
Fixed by applying patches like this:
```
--- g2o/types/sclam2d/CMakeLists.txt	2022-02-21 10:25:57.048311896 +0300
+++ g2o/types/sclam2d/CMakeLists.txt-new	2022-02-21 10:26:37.764747755 +0300
@@ -12,9 +12,9 @@
-if(G2O_HAVE_OPENGL)
-  target_link_libraries(types_sclam2d ${OPENGL_gl_LIBRARY})
-endif()
+#if(G2O_HAVE_OPENGL)
+#  target_link_libraries(types_sclam2d ${OPENGL_gl_LIBRARY})
+#endif()
```

RainerKuemmerle, 2022-02-23:
Is there more? I created a commit where we do not use that variable.

pddl, 2022-02-24:
Hi Rainer! 99% sure I have in-correct implementations of glvnd and mesa. Sorry and looks like your code is correct and no need to disable OPENGL_gl_LIBRARY. Once I've got g2o working in Waylanld, I'll let you know. Please close this issue.

closed 2022-02-24 by pddl