RainerKuemmerle/g2o #575: failed to add vertex?

aopaw, 2022-03-01:
Hello,

I tried using a simple example to optimize poses. I have a class encapsulates several g2o items to do this, and all of pointers of them are initialized in the constructor, see 
<code>
class Pose{

private:
	g2o::BlockSolver_6_3 * mog2oBS63;  // Solver type for pose matrix type.
	g2o::LinearSolverCSparse<g2o::BlockSolver_6_3::PoseMatrixType> * mog2oLSCSparse; // Linear solver.
	g2o::OptimizationAlgorithmLevenberg * mog2oOALeven;  // Levenberg optimizer.
	g2o::SparseOptimizer mog2oSOpt;
};
Pose::Pose(){
	// Allocating memory for g2o things.
	mog2oLSCSparse = new g2o::LinearSolverCSparse<g2o::BlockSolver_6_3::PoseMatrixType>();  // Linear solver initialization.
	mog2oLSCSparse->setBlockOrdering(false);
	mog2oBS63 = new g2o::BlockSolver_6_3(mog2oLSCSparse);  // Block solver initialization.
	mog2oOALeven = new g2o::OptimizationAlgorithmLevenberg(mog2oBS63); // Optimizer initialization.
	mog2oSOpt.setAlgorithm(mog2oOALeven);  // Set algorithm for optimizer.
	mog2oSOpt.setVerbose(false);
}
</code>

In main(), an object of Pose is declared and then a member function Run() of Pose is called to get everything working.
In Run(), I tried to add vertices into mog2oSOpt but failed to do that and don't know why. Here is a part of the code in Run()
<code>
Pose::Run(){
	g2o::VertexSE3 * pg2oVSE3 = new g2o::VertexSE3();
	pg2oVSE3->setId(0);
	pg2oVSE3->setEstimate(Eigen::Isometry3d::Identity());
	pg2oVSE3->setFixed(true);
	if (mog2oSOpt.addVertex(pg2oVSE3))
	{
		std::cout << "- vertex added: " << mog2oSOpt.vertices().size() << std::endl;
	}
	else
		std::cerr << "- vertex adding failed!" << std::endl;
}
</code>
The output is simply bizarre, see
- vertex added: 6888592

RainerKuemmerle, 2022-03-01:
Can you please provide a cpp with the full example.

aopaw, 2022-03-02:
> Can you please provide a cpp with the full example.

Hello, Rainer. The code is simply a copy of the main part in my project.
The object of class Pose is declared first in main() and then member function Run() is called and followed by a return 0.
```
int main(void)
{
       Pose oPose;
       oPose.Run();
       return 0;
}
```

Header files in pose.h.
```
#include <iostream>
#include <g2o/core/block_solver.h>
#include <g2o/core/optimization_algorithm_levenberg.h>
//#include <g2o/solvers/eigen/linear_solver_eigen.h>
#include <g2o/solvers/csparse/linear_solver_csparse.h>
#include <g2o/solvers/eigen/linear_solver_eigen.h>
#include <g2o/core/robust_kernel_factory.h>
#include <g2o/types/slam3d/types_slam3d.h>
#include <g2o/stuff/timeutil.h>

class Pose{
public:
           Pose();
           ~Pose();
public:
          void Run();

private:
	g2o::BlockSolver_6_3 * mog2oBS63;  // Solver type for pose matrix type.
	g2o::LinearSolverCSparse<g2o::BlockSolver_6_3::PoseMatrixType> * mog2oLSCSparse; // Linear solver.
	g2o::OptimizationAlgorithmLevenberg * mog2oOALeven;  // Levenberg optimizer.
	g2o::SparseOptimizer mog2oSOpt;
}
```

The pose.cpp file:
```
#include <pose.h>

Pose::Pose()
{
	// Allocating memory for g2o things.
	mog2oLSCSparse = new g2o::LinearSolverCSparse<g2o::BlockSolver_6_3::PoseMatrixType>();  // Linear solver initialization.
	mog2oLSCSparse->setBlockOrdering(false);
	mog2oBS63 = new g2o::BlockSolver_6_3(mog2oLSCSparse);  // Block solver ```
initialization.
	mog2oOALeven = new g2o::OptimizationAlgorithmLevenberg(mog2oBS63); // Optimizer initialization.
	mog2oSOpt.setAlgorithm(mog2oOALeven);  // Set algorithm for optimizer.
	mog2oSOpt.setVerbose(false);
}

Pose::~Pose()
{}

void Pose::Run()
{

	g2o::VertexSE3 * pg2oVSE3 = new g2o::VertexSE3();
	pg2oVSE3->setId(0);
	pg2oVSE3->setEstimate(Eigen::Isometry3d::Identity());
	pg2oVSE3->setFixed(true);
	if (mog2oSOpt.addVertex(pg2oVSE3))
	{
		std::cout << "- vertex added: " << mog2oSOpt.vertices().size() << std::endl;
	}
	else
		std::cerr << "- vertex adding failed!" << std::endl;
}
```
The main.cpp:
```
#include <pose.h>
int main(void)
{
       Pose oPose;
       oPose.Run();
       std::cin.ignore();
       return 0;
}
```

BTW, the g2o I am using is a version of 2016.

aopaw, 2022-03-02:
Tried using release version of 20170730, failed as well.

aopaw, 2022-03-02:
or this
```
#include <g2o/core/block_solver.h>
#include <g2o/core/optimization_algorithm_levenberg.h>
//#include <g2o/solvers/eigen/linear_solver_eigen.h>
#include <g2o/solvers/csparse/linear_solver_csparse.h>
#include <g2o/solvers/eigen/linear_solver_eigen.h>
#include <g2o/core/robust_kernel_factory.h>
#include <g2o/types/slam3d/types_slam3d.h>
#include <g2o/stuff/timeutil.h>
int main(void)
{
	g2o::BlockSolver_6_3::LinearSolverType* mog2oLSCSparse = new g2o::LinearSolverCSparse<g2o::BlockSolver_6_3::PoseMatrixType>();
	g2o::BlockSolver_6_3 *	mog2oBS63 = new g2o::BlockSolver_6_3(mog2oLSCSparse);
	g2o::OptimizationAlgorithmLevenberg* mog2oOALeven = new g2o::OptimizationAlgorithmLevenberg(mog2oBS63); // Optimizer initialization.
	g2o::SparseOptimizer mog2oSOpt;
	mog2oSOpt.setAlgorithm(mog2oOALeven);  // Set algorithm for optimizer.
	mog2oSOpt.setVerbose(false);

	g2o::VertexSE3 * pg2oVSE3 = new g2o::VertexSE3();
	pg2oVSE3->setId(0);
	pg2oVSE3->setEstimate(g2o::SE3Quat());
	pg2oVSE3->setFixed(true);
	if (mog2oSOpt.addVertex(pg2oVSE3))
	{
		std::cout << "- vertex added: " << mog2oSOpt.vertices().size() << std::endl;
	}
	else
		std::cerr << "- vertex adding failed!" << std::endl;

	std::cin.ignore();
	return 0;
}
```

aopaw, 2022-03-02:
@RainerKuemmerle Hi, Rainer. Can you please give me some help about this? Am I use it in a wrong way? I am not very good at C++.

RainerKuemmerle, 2022-03-02:
I added a branch based on 20170730 which contains your example. Works with GCC on Ubuntu. I cannot test with older MSVC on Windows. CI is using MSVC 2019.
```
$ ../bin/playground
- vertex added: 1
^C
```

aopaw, 2022-03-03:
> I added a branch based on 20170730 which contains your example. Works with GCC on Ubuntu. I cannot test with older MSVC on Windows. CI is using MSVC 2019.
> 
> ```
> $ ../bin/playground
> - vertex added: 1
> ^C
> ```

Oh, that s so touched. Thank you very much, Rainer. Anyway, I come to tell it works on VS2013 as soon as compiling it in Debug mode finished. I will try in my project.

aopaw, 2022-03-03:
Hi, @RainerKuemmerle. After doing test many times, it still fails to get a normal output. And finally, I create a new independent project without any other library. The code is still playground with which g2o and eigen3 (the same one for compiling g2o) are used as 3rd libraries. Then the result is still a bizarre number but an error of 'Access violation executing location' occurs at return 0. There is no this error while running the playground of example in g2o. Do you have any idea about this?

RainerKuemmerle, 2022-03-03:
One reason can be different compiler flags causing different behavior. In particular when march=native is defined for g2o but not on the project using g2o.

aopaw, 2022-03-04:
Hello, @RainerKuemmerle.
I got it. The error comes from a quite stupid mistake in my CMakeLists.txt.
The default link mode chooses Release while I did not give any instruction of CMAKE_BUILD_TYPE at the beginning in CMakeLists.txt, so the project opened with VS2013 is Debug mode in default. But static libs of Release version are linked to the target because I give an if-else at the end of the CMakeLists.txt. Then it goes to ELSE in case if(CMAKE BUILD TYPE) is false, i.e. it is not given at all. That really sucks wasting 2 days. I'd better say sorry :).

closed 2022-03-05 by RainerKuemmerle