RainerKuemmerle/g2o #576: problem in stuff/misc.h while building with vs2013

aopaw, 2022-03-02:
Hello, guys. 
I am now trying to use the up-to-date version but it seems my compiler or IDE component is out-to-date. Here is the error info

```
5>  Building Custom Rule F:/CPPs/slam14/ch7_VO1/source/Party3rd/g2o/source/g2o/examples/interactive_slam/slam_parser/parser/CMakeLists.txt
2>F:\CPPs\slam14\ch7_VO1\source\Party3rd\g2o\source\g2o/stuff/misc.h(68): error C2146: syntax error : missing ';' before identifier 'number_t'
2>F:\CPPs\slam14\ch7_VO1\source\Party3rd\g2o\source\g2o/stuff/misc.h(68): error C2433: 'constexpr' : 'inline' not permitted on data declarations
2>F:\CPPs\slam14\ch7_VO1\source\Party3rd\g2o\source\g2o/stuff/misc.h(68): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
2>F:\CPPs\slam14\ch7_VO1\source\Party3rd\g2o\source\g2o/stuff/misc.h(70): error C2146: syntax error : missing ';' before identifier 'number_t'
2>F:\CPPs\slam14\ch7_VO1\source\Party3rd\g2o\source\g2o/stuff/misc.h(70): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
2>F:\CPPs\slam14\ch7_VO1\source\Party3rd\g2o\source\g2o/stuff/misc.h(70): error C2086: 'int g2o::constexpr' : redefinition
2>          F:\CPPs\slam14\ch7_VO1\source\Party3rd\g2o\source\g2o/stuff/misc.h(68) : see declaration of 'g2o::constexpr'
```

aopaw, 2022-03-02:
https://stackoverflow.com/questions/20264644/constexpr-not-compiling-in-vc2013 
This one is not working, and I tried replacing it with const in misc.h but later found there are lots of this. So, give it up to stick on using vs2013.

aopaw, 2022-03-02:
BTW, in VS2013, #include <algorithm> is required for using std::max or min.

RainerKuemmerle, 2022-03-02:
> BTW, in VS2013, #include is required for using std::max or min.

Happy to receive PR to fix issues.
CI pipeline is using newer compiler Version.

RainerKuemmerle, 2022-03-02:
Looking at https://docs.microsoft.com/en-us/cpp/overview/visual-cpp-language-conformance?view=msvc-170 VS2013 does not fully support C++14 (which also sounds pretty natural). Hence, closing this here.

closed 2022-03-02 by RainerKuemmerle