RainerKuemmerle/g2o #577: failure in allocating memory for jacobian

aopaw, 2022-03-04:
One error after another.
```
- Time elapsed in total: 199.002
- 265 keyframes in total.
- Verticies in total: 265
- Edges in total: 264
Assertion failed: workspaceAllocated && "Error while allocating memory for the J
acobians", file F:\CPPs\slam14\g2o-f70\g2o\core\sparse_optimizer.cpp, line 213
```
```
void Pose::Optimize()
{
	std::cout << "- Verticies in total: "<< mog2oSOpt.vertices().size() << std::endl;
	std::cout << "- Edges in total: " << mog2oSOpt.edges().size() << std::endl;
	mog2oSOpt.setVerbose(true);
	mog2oSOpt.initializeOptimization();
	mog2oSOpt.optimize(10);
}
```
After adding a set of verticies and edges, I got another error in allocating memory in sparse_optimizer.cpp and failed to solve it. No idea again.

aopaw, 2022-03-05:
Hello, @RainerKuemmerle.
Am I using the code correct for adding edge in this way.

```
bool Pose::AddPose(const OneFrame & poOFL, const OneFrame & poOFR, const cv::Mat & poMRot, const cv::Mat & poMTrans)
{
	static g2o::RobustKernel * pg2oRKernel = g2o::RobustKernelFactory::instance()->construct("Cauchy");

	//std::cout << "- trans in addpose: " << poMTrans << std::endl;
	g2o::VertexSE3 * pg2oVSE3 = new g2o::VertexSE3();
	pg2oVSE3->setId(poOFR.miFid);
	pg2oVSE3->setEstimate(Eigen::Isometry3d::Identity());  // Set this to coming frames.
	mog2oSOpt.addVertex(pg2oVSE3);

	// For edge.
	g2o::EdgeSE3 * pg2oESE3 = new g2o::EdgeSE3();
	pg2oESE3->setId(poOFR.miFid);
	pg2oESE3->vertices().push_back(mog2oSOpt.vertex(poOFL.miFid));
	pg2oESE3->vertices().push_back(mog2oSOpt.vertex(poOFR.miFid));
	pg2oESE3->setRobustKernel(pg2oRKernel);

	// Info matrix.
	Eigen::Matrix<double, 6, 6> information = Eigen::Matrix< double, 6, 6 >::Identity();

	information(0, 0) = information(1, 1) = information(2, 2) = 100;
	information(3, 3) = information(4, 4) = information(5, 5) = 100;

	pg2oESE3->setInformation(information);

	///////////////////////////////////////////////////////////////////////
	Eigen::Matrix3d oEM3d;                // eigenMat.
	cv::cv2eigen(poMRot, oEM3d);          // Rotation matrix in types of Eigen.

	// Declare an eigen matrix which is an identity.
	Eigen::Isometry3d oEI3d = Eigen::Isometry3d::Identity();

	Eigen::AngleAxisd oEAAd(oEM3d);          // 3D rotation represented as an angle.

	oEI3d = oEAAd;
	oEI3d(0, 3) = poMTrans.at<double>(0, 0);        // Bind translation vector.
	oEI3d(1, 3) = poMTrans.at<double>(1, 0);
	oEI3d(2, 3) = poMTrans.at<double>(2, 0);
	pg2oESE3->setMeasurement(oEI3d.inverse());

	// Add edge.
	return(mog2oSOpt.addEdge(pg2oESE3));
}
```
The OneFrame is simply a customized class, in which the miFid is an integer assigning its index. 
According to the Debug breakpoint, in jacobian::allocate(), the maxnumverticies and another one are all -1. It passes all steps before this.

RainerKuemmerle, 2022-03-05:
Instead push_back, assign the vertices, see for example
https://github.com/RainerKuemmerle/g2o/blob/master/unit_test/slam3d/optimization_slam3d.cpp#L80,L81

aopaw, 2022-03-05:
> Instead push_back, assign the vertices, see for example https://github.com/RainerKuemmerle/g2o/blob/master/unit_test/slam3d/optimization_slam3d.cpp#L80,L81

Thank you very much. Working pretty well right now.

closed 2022-03-05 by RainerKuemmerle