RainerKuemmerle/g2o #578: After Successful optimize() from SparseOptimizer using CSparse : Segmentation fault (core dumped)

tallBearFromSeoul, 2022-03-04:
Hello,

I am trying to optimize keypoints obtained from frames of a video.
I run a separate algorithm of obtaining essential matrix of the scene with inlier keypoints obtained by matching the points between frames.
I reject once again the outliers such as the points that are behind the camera and reprojection error that's high through triangulation between two frame poses.
I get a pretty good estimation of the pose from the first two frames (almost no change in pose so the diagonal stays near 1, 1, 1).
Thus, I am expecting a good optimization from g2o by adding pose, vertex and edges according to the attached code below.
I have also attached the terminal output of the code.

However, for some reason, after the optimization runs successfully (50 steps), and the activeChi2() value is obtained, as soon as the function ends, there is segmentation fault.
I have made sure that it is not from other parts of my code by commenting this whole part from the program. Also, by printing before and after each operation to find out where exactly this error occurs.

[part_where_error_occurs.txt](https://github.com/RainerKuemmerle/g2o/files/8187207/part_where_error_occurs.txt)
[terminal_output.txt](https://github.com/RainerKuemmerle/g2o/files/8187209/terminal_output.txt)

part_where_error_occurs.txt : contains the two functions used for optimization
terminal_output.txt : contains the terminal output from running the whole code

To concisely explain the flow, 
in the optimize_map() function, first it runs the optimize() function, which runs successfully.
It even performs 50 steps of optimization successfully and even returns a value for opt->activeChi2(), which is at the end of optimize_map() function.
However, as soon as it comes back to optimize_map() function, it doesn't print the " cout<<"\n\n----isithere _ 1111----\n\n"; ", which is located right after the optimize_map() function call.
Thus, it is certain that the segmentation fault occurs in optimize() function.

Please help me out for this error.
I have no clue in solving it and decided to post an issue about it.

Thank you so much for your time for reading this.

I am running on Ubuntu 18.04 with cmake 3.22 gcc 8.4

Please let me know if you need more information.

Thank you again.

RainerKuemmerle, 2022-03-05:
Use gdb top get more insights of where the segmentation fault is happening. If you cannot figure something out, please provide a minimal working example.

tallBearFromSeoul, 2022-03-06:
Hello,

The following was the output from running gdb.
I am guessing the problem occurs when the Sparse Optimizer is calling destructor. 

Thread 1 "g2o_test" received signal SIGSEGV, Segmentation fault.
0x0000000000000021 in ?? ()
(gdb) where
#0  0x0000000000000021 in  ()
#1  0x00007ffff7b9408e in g2o::OptimizableGraph::Edge::~Edge() () at /usr/local/lib/libg2o_core.so
#2  0x00007ffff76ed6c1 in g2o::EdgeProjectXYZ2UV::~EdgeProjectXYZ2UV() () at /usr/local/lib/libg2o_types_sba.so
#3  0x00007ffff7b8bf8d in g2o::HyperGraph::clear() () at /usr/local/lib/libg2o_core.so
#4  0x00007ffff7b96cfe in g2o::OptimizableGraph::~OptimizableGraph() () at /usr/local/lib/libg2o_core.so
#5  0x00007ffff7bb1279 in g2o::SparseOptimizer::~SparseOptimizer() () at /usr/local/lib/libg2o_core.so
#6  0x000055555557146b in std::default_delete<g2o::SparseOptimizer>::operator()(g2o::SparseOptimizer*) const (this=<synthetic pointer>, __ptr=0x5555557ce7a0) at /usr/include/c++/8/bits/unique_ptr.h:350
#7  0x000055555557146b in std::unique_ptr<g2o::SparseOptimizer, std::default_delete<g2o::SparseOptimizer> >::~unique_ptr() (this=<synthetic pointer>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/unique_ptr.h:277
#8  0x000055555557146b in optimize(std::vector<Frame*, std::allocator<Frame*> > const&, std::set<Point*, std::less<Point*>, std::allocator<Point*> > const&, int const&, bool, int) (frames=..., points=..., local_window=<optimized out>, fix_points=true, rounds=50) at /home/gunny7828/Mar05_vSlam/src/optimize.cpp:18
#9  0x000055555556fb42 in Map::optimize_map(int const&, bool, int) (this=0x7fffffffc598, local_window=@0x7fffffffbfe0: 1, fix_points=<optimized out>, rounds=50) at /home/gunny7828/Mar05_vSlam/src/map.cpp:37
#10 0x0000555555572f15 in SLAM::process_frame(cv::Mat const&) (this=0x7fffffffc540, _img=...) at /home/gunny7828/Mar05_vSlam/src/slam.cpp:57
#11 0x000055555555ccdc in main (argc=<optimized out>, argv=<optimized out>) at /home/gunny7828/Mar05_vSlam/src/main.cpp:34
#12 0x00007ffff21cec87 in __libc_start_main (main=0x55555555c990 <main>, argc=1, argv=0x7fffffffc798, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffc788) at ../csu/libc-start.c:310
#13 0x000055555555cdfa in _start () at /usr/local/include/eigen3/Eigen/src/Core/util/IndexedViewHelper.h:125

RainerKuemmerle, 2022-03-06:
> #6 0x000055555557146b in std::default_deleteg2o::SparseOptimizer::operator()(g2o::SparseOptimizer*) const (this=, __ptr=0x5555557ce7a0) at /usr/include/c++/8/bits/unique_ptr.h:350
> #7 0x000055555557146b in std::unique_ptr<g2o::SparseOptimizer, std::default_deleteg2o::SparseOptimizer >::~unique_ptr() (this=, __in_chrg=) at /usr/include/c++/8/bits/unique_ptr.h:277
> #8 0x000055555557146b in optimize(std::vector<Frame*, std::allocator<Frame*> > const&, std::set<Point*, std::less<Point*>, std::allocator<Point*> > const&, int const&, bool, int) (frames=..., points=..., local_window=, fix_points=true, rounds=50) at /home/gunny7828/Mar05_vSlam/src/optimize.cpp:18
> #9 0x000055555556fb42 in Map::optimize_map(int const&, bool, int) (this=0x7fffffffc598, ***@***.***: 1, fix_points=, rounds=50) at /home/gunny7828/Mar05_vSlam/src/map.cpp:37
> #11 0x000055555555ccdc in main (argc=, argv=) at /home/gunny7828/Mar05_vSlam/src/main.cpp:34
> , argc=1, argv=0x7fffffffc798, init=, fini=, rtld_fini=, stack_end=0x7fffffffc788) at ../csu/libc-start.c:310

It seems like some parts are compiled in debug and some in release. Please make sure to have same compiler flags on both. that might cause issues.
If it still fails please provide stack trace with g2o also compiled in debug.
Which version of g2o?

tallBearFromSeoul, 2022-03-08:
Hello,

I apologize for the late reply.
It is indeed the latest version of g2o that I compiled in Release few days ago.
And I have built this project with RelWithDebInfo for more gdb output.
When I build it with Release flag, then the following is the output from gdb.
I will stay alert for the notification.
It takes a while for me to compile g2o again with Debug flag, could you take a look at this output first.
Then I will go ahead and compile g2o again with Debug along with the project in Debug as well.
Thank you again.

Thread 1 "g2o_test" received signal SIGSEGV, Segmentation fault.
0x0000000000000021 in ?? ()
(gdb) where
#0  0x0000000000000021 in  ()
#1  0x00007ffff7b9408e in g2o::OptimizableGraph::Edge::~Edge() () at /usr/local/lib/libg2o_core.so
#2  0x00007ffff76ed6c1 in g2o::EdgeProjectXYZ2UV::~EdgeProjectXYZ2UV() () at /usr/local/lib/libg2o_types_sba.so
#3  0x00007ffff7b8bf8d in g2o::HyperGraph::clear() () at /usr/local/lib/libg2o_core.so
#4  0x00007ffff7b96cfe in g2o::OptimizableGraph::~OptimizableGraph() () at /usr/local/lib/libg2o_core.so
#5  0x000055555557b13b in Map::optimize_map(int const&, bool, int, bool) ()
#6  0x000055555558a880 in SLAM::process_frame(cv::Mat const&) ()
#7  0x0000555555560067 in main ()

tallBearFromSeoul, 2022-03-09:
I realized that the problem was in defining the edge's Robust Kernel Huber.
Edge required a brand new kernel for each edge. 
So, when I only instantiated one kernel at the top and reused it that was the problem.
I am closing the issue as the problem is solved now.
Thank you for your support.

closed 2022-03-09 by tallBearFromSeoul