RainerKuemmerle/g2o #579: Internal Compiler Error In Jetson Xavier NX

junf1024, 2022-03-18:
Platform：Jetson Xavier NX
G2O：Latest
GCC：7.5.0
Eigen：3.3.4
When I compile directly, I get the following error

[ 26%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/star.cpp.o
**c++: internal compiler error: Segmentation fault signal terminated program cc1plus**
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
g2o/types/slam3d/CMakeFiles/types_slam3d.dir/build.make:134: recipe for target 'g2o/types/slam3d/CMakeFiles/types_slam3d.dir/vertex_se3.cpp.o' failed
make[2]: *** [g2o/types/slam3d/CMakeFiles/types_slam3d.dir/vertex_se3.cpp.o] Error 4
CMakeFiles/Makefile2:1103: recipe for target 'g2o/types/slam3d/CMakeFiles/types_slam3d.dir/all' failed
make[1]: *** [g2o/types/slam3d/CMakeFiles/types_slam3d.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 26%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/edge_types_cost_function.cpp.o

I have tried the following：
1. View current resources，and there is still a lot of free space in memory
2. Replacing the GCC version, such as 7, 8, 9, is the same failure phenomenon
3. Downgrading Eigen version to 3.2.10 still fails
4. In the same software environment (GCC、Eigen、CMake), TX2 compiles normally

Has anyone successfully compiled G2O under NX?

RainerKuemmerle, 2022-03-19:
Does it fail independent of the type of build, ie. Debug vs RelWithDebInfo vs Release.
I do not have access to an Nvidia Jetson. Is there a CI pipeline that offers access?

junf1024, 2022-03-25:
> Does it fail independent of the type of build, ie. Debug vs RelWithDebInfo vs Release. I do not have access to an Nvidia Jetson. Is there a CI pipeline that offers access?

Thanks for your reply.
I have tried both in Debug and Release mode and got the same failure result.

junf1024, 2022-04-24:
That may be an Nvidia Kit Version issue, not G2O...
According to my tests, there are the following results:
<html>
<body>
<!DOCTYPE html>

Seq. | Dev. | L4T Version | result
-- | -- | -- | --
1 | Jetson TX2 | 32.4.2 | Succ.
2 | Jetson Xavier NX | 32.6.1 | Failed.
3 | Jetson Xavier NX | 32.5.1 | Failed.
4 | Jetson AGX Xavier | 32.2.1 | Succ.
5 | Jetson AGX Xavier | 32.6.1 | Failed.
6 | Jetson AGX Xavier | 32.5.0 | Succ.
7 | Jetson Xavier NX | 32.7.1 | Succ.

</body>
</html>

So, when you choose NVIDIA NX as device, you should choose kit version 32.7.1 (for 2022.04)
The exact reason is unknown, but the above problems can be avoided by updating the system suite version. 
This question will be closed~

closed 2022-04-24 by junf1024