RainerKuemmerle/g2o #580: Cholesky failure when setting information for EdgeSE3

Victor-YG, 2022-03-19:
Hi guys, I am practising with g2o and I ran into the following issue:

I use VertexSE3 and EdgeSE3 for optimizing a pose graph (attached pose_graph.txt). When I don't set information, the internal matrix should be identity. In this case, the optimizer run smoothly. 

When I set the information for all edges to be identity manually, I got the Cholesky failure during optimization.

Cholesky failure, writing debug.txt (Hessian loadable by Octave)
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
iteration= 0     chi2= 11949.512891      time= 5.56399   cumTime= 5.56399        edges= 2685     schur= 0        lambda= 27694.822615    levenbergIter= 7
iteration= 1     chi2= 9076.688888       time= 0.146068  cumTime= 5.71006        edges= 2685     schur= 0        lambda= 18463.215077    levenbergIter= 1
iteration= 2     chi2= 5701.857555       time= 0.156099  cumTime= 5.86616        edges= 2685     schur= 0        lambda= 12308.810051    levenbergIter= 1
iteration= 3     chi2= 2460.821214       time= 0.1441    cumTime= 6.01026        edges= 2685     schur= 0        lambda= 8205.873367     levenbergIter= 1
iteration= 4     chi2= 188.476844        time= 0.145212  cumTime= 6.15547        edges= 2685     schur= 0        lambda= 5470.582245     levenbergIter= 1
iteration= 5     chi2= -770.435433       time= 0.146026  cumTime= 6.3015         edges= 2685     schur= 0        lambda= 3647.054830     levenbergIter= 1
iteration= 6     chi2= -826.579583       time= 0.147193  cumTime= 6.44869        edges= 2685     schur= 0        lambda= 2431.369887     levenbergIter= 1
iteration= 7     chi2= -826.579583       time= 1.37172   cumTime= 7.82042        edges= 2685     schur= 0        lambda= 87599332124038545408.000000     levenbergIter= 10

Please help me with this as I have got stuck for weeks. Thanks! Code attached.
[code.zip](https://github.com/RainerKuemmerle/g2o/files/8308430/code.zip)

Victor-YG, 2022-03-19:
The issue is fixed when I removed march=native in CMakeList.txt. It would nice if someone can explain why. Thanks.

RainerKuemmerle, 2022-03-19:
You need to have this flag also in your build, otherwise Eigen and header code in general might see different flags. And march-native seems to be a critical flag.
Default on master should be off for this flag due to this reason.
You example code worked for me.

Victor-YG, 2022-03-19:
So what should I do to fix the issue other than removing march=native?

Victor-YG, 2022-03-19:
It also appear to me that remove that flag also fixed the double free issue.

double free or corruption (out)
Aborted (core dumped)

RainerKuemmerle, 2022-03-19:
see for example some info on: https://wiki.gentoo.org/wiki/GCC_optimization#-march
If you do not want to tune/benchmark things just leave it out. If you want to have it, it has to be consistent between the g2o build and your example build.

closed 2022-03-19 by Victor-YG