RainerKuemmerle/g2o #581: Windows Windows dependencies install failed via 'script\install-deps-windows.bat'

Shortz79, 2022-03-22:
### **Issue:**
Dependencies install failed via 'script\install-deps-windows.bat' according to the vcpkg.
### **Test environment:**
OS: Windows
G2O version: Latest
vcpkg version: Latest with commit '6b6a35335c9064fac2ec6684c4f79808dce59b7d'

I try to build g2o from source by using the the vcpkg to install the dependencies. However it seems having some package conflict form these dependencies.
It's there any vcpkg commit were already been tested and able to build your repo. It will be great if you can provide a vcpkg commit(version tag) which is buildable.  

### **Output:**
`Starting package 11/14: lapack-reference:x64-windows
Building package lapack-reference[blas-select,core,noblas]:x64-windows...
CMake Error at ports/lapack-reference/portfile.cmake:8 (message):
  Can't build lapack-reference if clapack is installed.  Please remove
  clapack:x64-windows, and try to install lapack-reference:x64-windows again.
Call Stack (most recent call first):
  scripts/ports.cmake:145 (include)

Error: Building package lapack-reference:x64-windows failed with: BUILD_FAILED
Please ensure you're using the latest portfiles with `git pull` and `.\vcpkg update`.
Then check for known issues at:
  https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+lapack-reference
You can submit a new issue at:
  https://github.com/microsoft/vcpkg/issues/new?template=report-package-build-failure.md&title=[lapack-reference]+Build+error
including:
  package: lapack-reference[blas-select,core,noblas]:x64-windows -> 3.10.0
    vcpkg-tool version: 2022-03-09-1affd32f93b299d5a907816c328ca3ededb73a7e
    vcpkg-scripts version: 6b6a35335 2022-03-21 (2 hours ago)

Additionally, attach any relevant sections from the log files above.`

RainerKuemmerle, 2022-03-25:
Im not actively using Windows. It might be not working, this bat file.
`vcpkg --triplet x64-windows install eigen3`
Would be the minimal requirement to compile. See also https://github.com/RainerKuemmerle/g2o/blob/master/.github/workflows/windows.yml

lzyplayer, 2022-09-09:
I have come across the same problem as @Shortz79 . I have tried to remove `clapack` and install `lapack-reference`.  It works well! However, I cannot install Clapack after that because of conflicts.
```
PS D:\Software\vcpkg> ./vcpkg install eigen3 suitesparse clapack openblas ceres --triplet x64-windows
Computing installation plan...
The following packages are already installed:
    ceres[core]:x64-windows -> 2.1.0#2
    eigen3[core]:x64-windows -> 3.4.0#2
    openblas[core]:x64-windows -> 0.3.21
The following packages will be built and installed:
    clapack[core]:x64-windows -> 3.2.1#20
  * lapack[core]:x64-windows -> 2022-02-22
  * metis[core]:x64-windows -> 2022-07-27
    suitesparse[core]:x64-windows -> 5.8.0#1
Additional packages (*) will be modified to complete this operation.
ceres:x64-windows is already installed
eigen3:x64-windows is already installed
openblas:x64-windows is already installed
Detecting compiler hash for triplet x64-windows...
Restored 1 package(s) from C:\Users\blakelin\AppData\Local\vcpkg\archives in 241.3 ms. Use --debug to see more details.
Installing 1/4 clapack:x64-windows...
The following files are already installed in D:/Software/vcpkg/installed/x64-windows and are in conflict with clapack:x64-windows

Installed by lapack-reference:x64-windows
    lib/lapack.lib
    share/lapack/FindLAPACK.cmake
    share/lapack/vcpkg-cmake-wrapper.cmake

Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
    https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+clapack
You can submit a new issue at:
    https://github.com/microsoft/vcpkg/issues/new?template=report-package-build-failure.md&title=[clapack]+Build+error
Include '[clapack] Build error' in your bug report title, the following version information in your bug description, and attach any relevant failure logs from above.
    vcpkg-tool version: 2022-09-01-dfb82802c8cc562ce3b665a904a65b22314de724
    vcpkg-scripts version: 927006b24 2022-09-02 (3 days ago)
```

RainerKuemmerle, 2023-03-03:
Realized that this is the same as in https://github.com/RainerKuemmerle/g2o/issues/638
Is there any known way to install suitesparse with vcpkg that works at the moment? I would appreciate feedback from Windows users.

closed 2025-01-28 by github-actions[bot]