RainerKuemmerle/g2o #582: on-class type and does not name a type build error on Ubuntu 20.04

fvillml, 2022-03-25:
Hello, I cannot build g2o master branch on Ubuntu 20.04. I tried with g++ and gcc 10 and 9 and none work. Here is the result of my cmake:

```bash
$ cmake ..
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling on Unix
-- Found Cholmod: /usr/include/suitesparse  
-- A cache variable, namely BLAS_DIR, has been set to specify the install directory of BLAS
-- Looking for BLAS -- mkl.h not found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for MKL BLAS: not found
-- Looking for Goto BLAS: not found
-- Looking for Open BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Atlas BLAS: not found
-- Looking for PhiPACK BLAS: not found
-- Looking for CXML BLAS: not found
-- Looking for DXML BLAS: not found
-- Looking for SunPerf BLAS: not found
-- Looking for SCSL BLAS: not found
-- Looking for SGIMATH BLAS: not found
-- Looking for IBM ESSL BLAS: not found
-- Looking for IBM ESSL MT BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for Apple BLAS: not found
-- Looking for NAS BLAS: not found
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- Looking for Generic BLAS: found
-- A library with BLAS API found.
-- BLAS_LIBRARIES /usr/lib/x86_64-linux-gnu/libblas.so
-- Looking for BLAS -- mkl.h not found
-- Looking for MKL BLAS: not found
-- Looking for Goto BLAS: not found
-- Looking for Open BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Atlas BLAS: not found
-- Looking for PhiPACK BLAS: not found
-- Looking for CXML BLAS: not found
-- Looking for DXML BLAS: not found
-- Looking for SunPerf BLAS: not found
-- Looking for SCSL BLAS: not found
-- Looking for SGIMATH BLAS: not found
-- Looking for IBM ESSL BLAS: not found
-- Looking for IBM ESSL MT BLAS: not found
[... 157 lines cut ...]
      |     ^~~~~~~~~
/home/user/g2o/g2o/solvers/csparse/csparse_extension.cpp:56:5: error: 'cs_ltsolve' was not declared in this scope
   56 |     cs_ltsolve(N->L, x);        /* x = L'\x */
      |     ^~~~~~~~~~
/home/user/g2o/g2o/solvers/csparse/csparse_extension.cpp:57:16: error: request for member 'pinv' in '* S', which is of non-class type 'const int'
   57 |     cs_pvec(S->pinv, x, b, n);  /* b = P'*x */
      |                ^~~~
/home/user/g2o/g2o/solvers/csparse/csparse_extension.cpp:57:5: error: 'cs_pvec' was not declared in this scope
   57 |     cs_pvec(S->pinv, x, b, n);  /* b = P'*x */
      |     ^~~~~~~
/home/user/g2o/g2o/solvers/csparse/csparse_extension.cpp:59:3: error: 'cs_nfree' was not declared in this scope; did you mean 'c_free'?
   59 |   cs_nfree(N);
      |   ^~~~~~~~
      |   c_free
/home/user/g2o/g2o/solvers/csparse/csparse_extension.cpp: At global scope:
/home/user/g2o/g2o/solvers/csparse/csparse_extension.cpp:68:1: error: 'csn' does not name a type; did you mean 'csc'?
   68 | csn *cs_chol_workspace(const cs *A, const css *S, int *cin, number_t *xin) {
      | ^~~
      | csc
make[2]: *** [g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/build.make:76: g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/csparse_extension.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1962: g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Perhaps I'm missing something for compilation?

fvillml, 2022-03-25:
Using 
```
cmake -DG2O_USE_CSPARSE=OFF ..
```
works, what are the drawbacs of not using CSPARSE?

fvillml, 2022-03-25:
> 

Actually I need CSPARSE. I'm trying to compile `teb_local_planner` from source and it spawns an error if g2o compiled without CSPARSE

fvillml, 2022-03-25:
> Using
> 
> ```
> cmake -DG2O_USE_CSPARSE=OFF ..
> ```
> 
> works, what are the drawbacs of not using CSPARSE?

Actually I need CSPARSE. I'm trying to compile teb_local_planner from source and it spawns an error if g2o compiled without CSPARSE

RainerKuemmerle, 2022-03-25:
Are you trying to compile master?
Do you have /usr/include/suitesparse/cs.h as your include for csparse? Or some other cs.h in your include path?
Can you please execute
` make VERBOSE=1 solver_csparse`
The CI (https://github.com/RainerKuemmerle/g2o/blob/master/.github/workflows/ci.yml) runs on focal and I also have one laptop on focal without any issue.

fvillml, 2022-03-25:
Yes I'm trying to compile master.

I do have the header
```bash
$ ls /usr/include/suitesparse/cs.h 
/usr/include/suitesparse/cs.h
```

Here the result with verbose
```bash
[ 28%] Building CXX object g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/csparse_helper.cpp.o
cd /home/user/g2o/build/g2o/solvers/csparse && /usr/bin/c++  -DUNIX -Dcsparse_extension_EXPORTS -I/home/user/g2o -I/home/user/g2o/build -I/home/user/g2o/g2o/solvers/csparse -isystem /usr/include/eigen3  -Wall -W -O3 -DNDEBUG -O3 -fPIC   -msse2 -msse3 -msse4.1 -msse4.2 -std=gnu++14 -o CMakeFiles/csparse_extension.dir/csparse_helper.cpp.o -c /home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp
In file included from /home/user/g2o/g2o/solvers/csparse/csparse_helper.h:30,
                 from /home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:27:
/home/user/g2o/g2o/solvers/csparse/csparse_extension.h:41:27: error: 'csn' does not name a type; did you mean 'csc'?
   41 | G2O_CSPARSE_EXTENSION_API csn* cs_chol_workspace(const cs* A, const css* S,
      |                           ^~~
      |                           csc
/home/user/g2o/g2o/solvers/csparse/csparse_extension.h:43:52: error: 'cs' does not name a type; did you mean 'csc'?
   43 | G2O_CSPARSE_EXTENSION_API int cs_cholsolsymb(const cs* A, number_t* b,
      |                                                    ^~
      |                                                    csc
/home/user/g2o/g2o/solvers/csparse/csparse_extension.h:44:52: error: 'css' does not name a type; did you mean 'csc'?
   44 |                                              const css* S, number_t* workspace,
      |                                                    ^~~
      |                                                    csc
In file included from /home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:27:
/home/user/g2o/g2o/solvers/csparse/csparse_helper.h:40:75: error: 'cs' does not name a type; did you mean 'csc'?
   40 | G2O_CSPARSE_EXTENSION_API bool writeCs2Octave(const char* filename, const cs* A,
      |                                                                           ^~
      |                                                                           csc
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:36:49: error: 'cs' does not name a type; did you mean 'csc'?
   36 | bool writeCs2Octave(const char* filename, const cs* A, bool upperTriangular) {
      |                                                 ^~
      |                                                 csc
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp: In function 'bool g2o::csparse_extension::writeCs2Octave(const char*, const int*, bool)':
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:37:17: error: request for member 'n' in '* A', which is of non-class type 'const int'
   37 |   int cols = A->n;
      |                 ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:38:17: error: request for member 'm' in '* A', which is of non-class type 'const int'
   38 |   int rows = A->m;
      |                 ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:40:10: error: request for member 'nz' in '* A', which is of non-class type 'const int'
   40 |   if (A->nz == -1) {  // CCS matrix
      |          ^~
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:41:24: error: request for member 'p' in '* A', which is of non-class type 'const int'
   41 |     const int* Ap = A->p;
      |                        ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:42:24: error: request for member 'i' in '* A', which is of non-class type 'const int'
   42 |     const int* Ai = A->i;
      |                        ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:43:29: error: request for member 'x' in '* A', which is of non-class type 'const int'
   43 |     const number_t* Ax = A->x;
      |                             ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:46:24: error: request for member 'p' in '* A', which is of non-class type 'const int'
   46 |     const int* Aj = A->p;
      |                        ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:47:24: error: request for member 'i' in '* A', which is of non-class type 'const int'
   47 |     const int* Ai = A->i;
      |                        ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:48:29: error: request for member 'x' in '* A', which is of non-class type 'const int'
   48 |     const number_t* Ax = A->x;
      |                             ^
/home/user/g2o/g2o/solvers/csparse/csparse_helper.cpp:49:44: error: request for member 'nz' in '* A', which is of non-class type 'const int'
   49 |     return writeTripletMatrix(filename, A->nz, rows, cols, Ai, Aj, Ax,
      |                                            ^~
make[3]: *** [g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/build.make:63: g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/csparse_helper.cpp.o] Error 1
make[3]: Leaving directory '/home/user/g2o/build'
make[2]: *** [CMakeFiles/Makefile2:2067: g2o/solvers/csparse/CMakeFiles/csparse_extension.dir/all] Error 2
make[2]: Leaving directory '/home/user/g2o/build'
make[1]: *** [CMakeFiles/Makefile2:2047: g2o/solvers/csparse/CMakeFiles/solver_csparse.dir/rule] Error 2
make[1]: Leaving directory '/home/user/g2o/build'
make: *** [Makefile:550: solver_csparse] Error 2
```

fvillml, 2022-03-25:
I realized that I have another header called cs.h in `/usr/local/include/` from osqp library. Apparently from the cmake output it finds this one first so I get
```bash
Found CSparse: /usr/local/include 
```
instead of 
```bash
Found CSparse: /usr/include/suiteparse
```
Doing 
```bash
cmake -DCSPARSE_INCLUDE_DIR=/usr/include/suitesparse/ ..
```
works well, thank you !

closed 2022-03-25 by fvillml